MangoAutomation/ma-core-public #1150: Data Point CSV Export - useful fields missing

Puckfist, 2017-11-20:
It has been a while since the CSV export for data points was updated. Perhaps we should do this all at once? Potential columns that could be added to the base model...

[ ] Prevent extreme sets
[ ] Prevent extreme set high limit
[ ] Prevent extreme set low limit
[ ] Rollup
[ ] Path

We probably can't do tags in the CSV, right? Or, would we just specify a format? I've not attempted to make this an exhaustive list of possibilities yet.

jazdw, 2017-11-20:
@terrypacker and I discussed what to do about CSV in the v2 controller and we decided that we would come up with a new solution so the v2 controller doesn't have CSV yet. Its probably not hard to add some extra columns to the v1 models. It wont be possible to add tags though.

When we do implement CSV for the v2 controller we should be able to get tags working using column headers like "tags.siteName" etc.

terrypacker, 2017-12-12:
See #1172

terrypacker, 2018-07-16:
Closing as v2 CSV has been implemented and released in Mango 3.4

closed 2018-07-16 by terrypacker