MangoAutomation/ma-core-public #1151: Event type permissions - improper event subset for nonadmin users

Puckfist, 2017-11-20:
Lower privilege users will only have events from data sources and data points they have permissions on added to the user events for that user. Therefore, they would miss scheduled events, advanced scheduled events, maintenance events, system events, potentially audit events, etc

A relevant method is Permissions.hasEventTypePermission

terrypacker, 2018-04-04:
This will require some discussion but we should probably create permissions for all event types.  This will involve adding the permissionTypeNames to the EventTypeDefinition, then provide a method in the definition to check the permissions.  This check would be at the bottom of the Permission.hasEventTypePermission method.

One thing to discuss is if there are multiple permission types such as read and edit or is edit an admin only permission in which case the permissions would only be checked for read/view of a user.

terrypacker, 2018-04-11:
I added the framework and kept things as they were except for AdvancedScheduler Events, they now are viewable based on the ScheduleVO's permissions.

closed 2018-04-16 by Puckfist