MangoAutomation/ma-core-public #1155: JSONEmportScriptUtility - Deadlock with ImportTask

terrypacker, 2017-11-27:
When importing a Scripting data source, with points, and with its script creating data point deadlock can occur:

1.  ImportTask --> DataPointImporter --> RuntimeManagerImpl.startDataPoint() --> ScriptDataSourceRT.addDataPoint()

2. ScriptDataSourceRT.execute() --> JsonEmportScriptUtility.doImport() --> DataPointImporter --> RuntimeManagerImpl.saveDataPoint() --> RuntimeManagerImpl.stopDataPoint(int)

[image]
[image]

terrypacker, 2017-11-28:
The code to fix this is in the private modules repo on the main-scripting-import branch.

Notes for when we get to this:

1.  I think we can get rid of the execution lock if we are going to allow changing the context while the script runs.  It is only there to keep the changes out of the context.  So I figure if it is acceptable to do to itself it should be acceptable for anyone to modify the context?  No matter what the use of it should be removed from the terminate method.

2.  We are not removing the point as a listener in the removeDataPoint method.  Any reason why?  We should add code for that I think.

3.  I think there is a potential problem when creating data points from a data source while it is enabled AND importing the JSON for the points.  I can see how duplicate points could get added to the runtime in this scenario and it probably has to do with this issue:  https://github.com/infiniteautomation/ma-core-public/issues/1123  While I’m importing the JSON the script starts running and creating points at the same time.  I assume it is creating some/all of the same points that are in the JSON.

Puckfist, 2017-12-11:
2) Add and remove data points are only applicable for points on the data source. The LocalPointListener removes itself at point terminated

closed 2018-04-12 by Puckfist