MangoAutomation/ma-core-public #1161: Add more indexes on events table

jazdw, 2017-12-05:
This follows on from https://github.com/infiniteautomation/ma-dashboards/issues/111

This needs to happen sooner rather than later. We currently are querying for active events for data points on watchlists and on a lot of dashboards etc using a REST query like this -
/rest/v1/events?dataPointId=59&active=true&limit(0,0)

We need to investigate what indexes to add to make this faster. From my previous comments on the other issue I suspect that we need to add indexes on
- typeRef1
- (rtnApplicable, rtnTs)

terrypacker, 2017-12-14:
A large part of this task will be determining what indexes we need.  Perhaps compile a list of the common queries by patrolling the UI pages and watching the XHR requests.

terrypacker, 2017-12-20:
We could improve things further because we seem to be making queries like this:

SELECT evt.id,evt.typeName,evt.subtypeName,evt.typeRef1,evt.typeRef2,evt.activeTs,evt.rtnApplicable,evt.rtnTs,evt.rtnCause,evt.alarmLevel,evt.message,evt.ackTs,evt.ackUserId,evt.alternateAckSource,u.username,(select count(1) from userComments where commentType=1 and typeKey=evt.id) as cnt ,ue.silenced FROM events  AS evt  LEFT JOIN  users u  ON evt.ackUserId = u.id LEFT JOIN  userEvents ue  ON evt.id=ue.eventId WHERE  (  typeName = 'DATA_POINT' AND typeRef1 = ?  AND evt.rtnTs IS ?  AND evt.rtnApplicable=?  AND ue.userId=?  ) LIMIT ? 
Args: [42, null, Y, 1, 0]]  

With a limit of 0.  I believe this is because all the  UI wants is a count but there is no way to do only a count API request.  We could build in some logic that if the limit is 0 we don't perform the select.

terrypacker, 2017-12-20:
I implemented the logic of skipping the SELECT query if the limit is 0 and I feel that the only index necessary is on typeRef1 from my testing.

terrypacker, 2017-12-20:
The queries that have been optimized are:

http://localhost:8080/rest/v1/events?eq(dataPointId,42)&eq(active,true)&limit(0)
http://localhost:8080/rest/v1/events?dataPointId=42&sort(-activeTimestamp)&limit(15,0)
http://localhost:8080/rest/v1/events?dataPointId=42&acknowledged=false&limit(0)

http://localhost:8080/rest/v1/events?eq(dataPointId,42)&eq(active,true)&limit(100)
http://localhost:8080/rest/v1/events?dataPointId=42&sort(-activeTimestamp)&limit(100,0)
http://localhost:8080/rest/v1/events?dataPointId=42&acknowledged=false&limit(100)

Down from over 2s to around 100ms on my test H2 database with:

Events Count: 1277772
Data Point Events Count: 277768
Data Point Count: 2778
Data Source Count: 28

terrypacker, 2017-12-20:
During testing I did notice that **periodically** when I go to the data point details page the following requests are problematic:

This request gets cancelled:
``` rest/v1/events/active-summary ```

This request takes 42s:
``` rest/v2/data-points/DP_9af20ac7-34cf-40bd-b747-5df3da06eae0 ```

Generally I can make this happen on any data point but I must leave the UI idle for some time and then refresh the data point details page.

terrypacker, 2017-12-20:
This is in Upgrade21 and the create tables sql files.

jazdw, 2017-12-20:
> With a limit of 0. I believe this is because all the UI wants is a count but there is no way to do only a count API request. We could build in some logic that if the limit is 0 we don't perform the select.

A select with a limit of 0 should complete immediately as far as I'm aware. Regardless, if there are proper indexes the query should complete in a timely manner.

>The queries that have been optimized are:

These queries and the ones you posted results for in the email are only for data point queries. We also need to add indexes for finding all events that are active, regardless of type.

>This request gets cancelled:
rest/v1/events/active-summary

This is exactly why we need the index on (rtnApplicable, rtnTs) as described in the initial issue comment.

>This request takes 42s:
rest/v2/data-points/DP_9af20ac7-34cf-40bd-b747-5df3da06eae0

I suspect the DB is overloaded trying to find the active events or something, this should never take 42s

I'm going to suggest you run these tests in the MySQL workbench outside of Mango. Get the SQL query to use from the log/console then paste it into MySQL. This will let you see the query times but more importantly you can use the visual explain tool to see what indexes are being used and the "performance cost" of each query.

Note, make sure query_cache_type is set to 0 in your options file (Performance tab in MySQL workbench).

terrypacker, 2017-12-20:
There is a basic issue with any query that uses the EVENT_SELECT_WITH_USER_DATA query base in the EventDao, this query is very resource intensive.  We will need to look at how we can achieve what it is being used for in another way after the 3.3.0 release.  Basically selecting events for a user by joining to the User and UserEvents table can be a very large query.

terrypacker, 2017-12-20:
``` rest/v1/events/active-summary ```

This query is very slow and fills the user event cache with all un-silenced events but all we really need are the totals and latest event for every alarm level.  We can do that with this query:

```
SELECT * FROM
(
SELECT events.alarmLevel, COUNT(events.id), MAX(events.activeTs) AS maxTs
from userEvents
LEFT JOIN events ON events.id=userEvents.eventId
WHERE  userEvents.userId=1 AND userEvents.silenced='N'
GROUP BY events.alarmLevel
) AS agg
LEFT JOIN events ON agg.maxTs = events.activeTs AND agg.alarmLevel = events.alarmLevel
GROUP BY events.alarmLevel
```

terrypacker, 2017-12-21:
Here is an example EventDao method that could return the summaries.  If we wire this information into the UserEventCache we may get better performance.  Currently the way the cache works is more efficient than making this query over and over.  Filling the cache does take time and requires more memory to hold all events, but after it is full the performance benefit greatly outweighs using this query at every page request.  Basically this query is faster than the one used in EventDao.getAllUnsilencedEvents(userId) but would need to be called many times where the cache implementation currently only needs to make 1 call to the database each time a user logs in with an empty cache.

```
    /**
     * 
     * Summary of all the unsilenced events for a user, return in a map Level -> Summary
     * @param userId
     * @return
     */
    public Map<Integer, UserEventLevelSummary> getUnsilencedUserEventSummaries(int userId){
        StringBuilder query = new StringBuilder();
        query.append("SELECT agg.alarmLevel, agg.total, events.id");
        query.append(" FROM (");
        query.append("SELECT events.alarmLevel, COUNT(events.id) as total, MAX(events.activeTs) AS maxTs ");
        query.append("from userEvents ");
        query.append("LEFT JOIN events ON events.id=userEvents.eventId ");
        query.append("WHERE  userEvents.userId=? AND userEvents.silenced='N' ");
        query.append("GROUP BY events.alarmLevel) ");
        query.append("AS agg ");
        query.append("LEFT JOIN events ON agg.maxTs = events.activeTs AND agg.alarmLevel = events.alarmLevel ");
        query.append("GROUP BY events.alarmLevel");
        return query(query.toString(), new Object[] {userId}, new ResultSetExtractor<Map<Integer, UserEventLevelSummary>>() {

            @Override
            public Map<Integer, UserEventLevelSummary> extractData(ResultSet rs)
                    throws SQLException, DataAccessException {
                Map<Integer, UserEventLevelSummary> alarmCounts = new HashMap<>();
                alarmCounts.put(AlarmLevels.LIFE_SAFETY, new UserEventLevelSummary(AlarmLevels.LIFE_SAFETY));
                alarmCounts.put(AlarmLevels.CRITICAL, new UserEventLevelSummary(AlarmLevels.CRITICAL));
                alarmCounts.put(AlarmLevels.URGENT, new UserEventLevelSummary(AlarmLevels.URGENT));
                alarmCounts.put(AlarmLevels.WARNING, new UserEventLevelSummary(AlarmLevels.WARNING));
                alarmCounts.put(AlarmLevels.IMPORTANT, new UserEventLevelSummary(AlarmLevels.IMPORTANT));
                alarmCounts.put(AlarmLevels.INFORMATION, new UserEventLevelSummary(AlarmLevels.INFORMATION));
                alarmCounts.put(AlarmLevels.NONE, new UserEventLevelSummary(AlarmLevels.NONE));
                alarmCounts.put(AlarmLevels.DO_NOT_LOG, new UserEventLevelSummary(AlarmLevels.DO_NOT_LOG));
                
                while(rs.next()) {
                    UserEventLevelSummary summary = alarmCounts.get(rs.getInt(1));
                    summary.setUnsilencedCount(rs.getInt(2));
                    summary.setLatest(getEventInstance(rs.getInt(3)));
                }
                return alarmCounts;
            }
        });
    }
```

terrypacker, 2017-12-21:
An additional improvement to the cache could be to load all the events into Memory only 1 time and reference them in a similar manner to the concept of the UserEvents table.  Currently the cache holds a list of all events applicable to a user for each user that is logged in.  This means that there are duplicate events held in memory, which is un-necessary.

terrypacker, 2017-12-21:
After further consideration I think it might be better to use the slower 'Summary' query because that only happens once on page load, and in the new UI this should only happen once during a user's session.

terrypacker, 2018-03-14:
Closing as we will be reworking the events system.

closed 2018-03-14 by terrypacker