MangoAutomation/ma-core-public #1173: Data Import - Deleting Latest Point does not clear it from Realtime Point Value Cache

terrypacker, 2017-12-12:
The RealTimeDataPointValueCache contains references to DataPointRTs when I deleted the latest value from Mango the DataPointRT.pointValue still retains a reference to this value...

We should ensure this doesn't happen in the DataImport module and the v2 REST controller's DELETE method.

terrypacker, 2018-01-10:
It appears that the purge method in the Runtime Manager calls DataPointRT.resetValues() which in turn calls PointValueCache.reset().  This doesn't actually reload any values from the database, it only trims the in memory cache to be the default size.

We probably want to reload the cache from the database at this point taking into consideration Interval logging ramifications.

Puckfist, 2018-02-23:
As simple as changing PointValueCache::reset to

public void reset() {
        List<PointValueTime> nc = dao.getLatestPointValues(dataPointId, defaultSize);
        maxSize = defaultSize;
        cache = nc;
    }

no?

Puckfist, 2018-02-23:
Interval logging tracks the necessary value in the DataPointRT

The only blemish I see on the simple solution is this is an extra database call for every point during the nightly purge, where the nightly purge is usually not going to delete the value in the cache. Perhaps passing in the time purged unto enables the cache to be smart about a db call

Puckfist, 2018-02-23:
The end delete time is now passed to a supporting resetValues() in the dprt that passes that time to the value cache. If the cache is smaller than the default size after that, it will refresh itself to the default size.

closed 2018-03-12 by Puckfist