MangoAutomation/ma-core-public #1186: NPE in ThreadModel

jazdw, 2017-12-19:
```
ERROR 2017-12-19T21:50:16,909 (com.serotonin.m2m2.web.mvc.rest.v1.exception.RestExceptionHandler.handleExceptionInternal:74) - null 
java.lang.NullPointerException: null
	at com.serotonin.m2m2.web.mvc.rest.v1.model.thread.ThreadModel.getLockInfo(ThreadModel.java:80) ~[mango-api-3.2.4.jar:?]
```

terrypacker, 2017-12-19:
The problem of the model was that it relied on ThreadInfo which is null for threads that no longer exist.

terrypacker, 2017-12-21:
To test we will need to create a bunch of very short lived threads and make a series of REST requests to rest/v1/threads as the only way this can happen is if the Thread dies between the lines of code that requests the list of ThreadInfos and the code that writes the JSON out of the REST controller.

Also should test ordering by some of the thread info properties that will be null:

- [ ] STATE
- [ ] LOCATION

closed 2018-01-18 by terrypacker