MangoAutomation/ma-core-public #1187: Lots of BACnet timeout messages in log file

Joelhaggar, 2017-12-20:
On BACnet systems I sometimes see 1000's of these messages blowing out the log files 

ERROR 2017-12-20T13:46:33,341 (com.serotonin.bacnet4j.util.PropertyUtils.requestPropertiesFromDevice:209) - Timeout while finding device 310089 
com.serotonin.bacnet4j.exception.BACnetTimeoutException: No response from instanceId 310089

I think this just means the device is offline and I'm not sure this needs to be a log message every time we poll or a log message at all.

Joelhaggar, 2017-12-20:
I'm seeing these ever millisecond and multiple per millisecond in some cases.

terrypacker, 2018-03-14:
To close this all that needs to be done is to release a new BACnet module with the updated BACnet4J library.  Will need to release BACnet4J too.

terrypacker, 2018-04-10:
Released BACnet4J 4.1.5 and now building module against it.  Should be no changes that effect the module but a retest is wise.