MangoAutomation/ma-core-public #1188: Persistent Publisher - Spams audit table

Puckfist, 2017-12-20:
The persistent TCP data source will create new records in the audit table from point connection even if nothing changed - especially obvious with points that have range renderers. I was testing with the publisher and the data source in the same instance.

Test config:
[publisher-audit-events.txt](https://github.com/infiniteautomation/ma-core-public/files/1577214/publisher-audit-events.txt)

terrypacker, 2017-12-20:
I think your test config is chopped off on the end.

Puckfist, 2017-12-20:
Really? I just redownloaded it, no issue.

It's just a virtual data source with two points (one numeric with a range renderer, one binary) being published from a tcp publisher to a data source in the same instance.

jazdw, 2017-12-20:
Probably something to do with the order of arrays then? How is the comparison done? Is it converted to JSON then the string compared?

I guess another question is why are points being saved if nothing changed...

terrypacker, 2017-12-20:
Yes I've been meaning to comment on that.  We made a similar change somewhere else so that the table wouldn't get spammed.
See this commit: 389122eb3f40ad3568eec7c4b25a89a5c9c0e8e1
And this issue: #891

terrypacker, 2017-12-21:
The JsonSerializableUtility does the heavy lifting of determining the changes between the Objects.  I can see some potential issues after reviewing the code as I think it should only focus on Sero JSON properties but it appears that the Map and Collections code also looks into the non-sero-json members.  The problem with that is there may be some runtime members set when the object is updated that don't actually represent a change to the object in the database.  

It will require a little more investigation though.

terrypacker, 2017-12-21:
Fixed the RangeRenderer bug it wasn't comparing the values for the ranges, it was doing an Object to Object comparison.  I've modified the difference detection to evaluate the differences down to the Object.class level correctly.

terrypacker, 2017-12-21:
Tested and no audit changes are detected when the publisher connects anymore, unless there really were changes.

closed 2017-12-21 by terrypacker