MangoAutomation/ma-core-public #1189: Persistent Sync - Array index out of bounds escaping to stderr

Puckfist, 2017-12-20:
Exception in thread "high-pool-2-thread-49" java.lang.IndexOutOfBoundsException: Index: 1, Size: 1
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.remove(ArrayList.java:492)
	at com.serotonin.m2m2.persistent.pub.SyncHandler$PointSync.run(SyncHandler.java:257)
	at com.serotonin.timer.sync.Synchronizer$TaskWrapper.run(Synchronizer.java:150)
	at com.serotonin.timer.Task.runTask(Task.java:179)
	at com.serotonin.timer.TaskWrapper.run(TaskWrapper.java:23)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

terrypacker, 2018-03-01:
I would fix this 2 ways.  1 by wrapping everything in the PointSync.run method in a try/catch and 2 by changing the list of points ```pointsToSync``` into a ConcurrentLinkedQueue or adding synchronization to that list.

terrypacker, 2018-03-01:
Released 3.3.1

closed 2018-03-01 by terrypacker