MangoAutomation/ma-core-public #1196: Incorrect useUnitAsSuffix updating a data point analog text renderer using REST

jazdw, 2018-01-08:
The useUnitAsSuffix setting gets overwritten if the point locator is set after the text renderer.

DataPointVO line 425
```
        // if data type changed from non numeric to numeric then use unit as suffix
        if (pointLocator.getDataTypeId() == DataTypes.NUMERIC
                && (this.pointLocator == null || this.pointLocator.getDataTypeId() != DataTypes.NUMERIC)
                && textRenderer instanceof ConvertingRenderer) {
            ConvertingRenderer cr = (ConvertingRenderer) textRenderer;
            cr.setUseUnitAsSuffix(true);
        }
```

terrypacker, 2018-01-11:
Its odd logic but I guess to keep the functionality we can just check to see if the pointLocator was null when we entered the method and if it was then set useUnitAsSuffix, otherwise don't touch it.

terrypacker, 2018-01-11:
Removed  code in c95e5789cfc9edb929ee4afbcde3541bd8e0c3ec

closed 2018-01-11 by terrypacker