MangoAutomation/ma-core-public #1203: Date Truncation - Consistency

terrypacker, 2018-01-18:
We need to settle on a general approach to truncating dates.  This happens differently in various places.

1.  Excel Reports - Truncates start to nearest rollup period (no end truncation)

2.  MangoApi - Point Value endpoints truncate the start and end dates.  (Truncating times to DAY that are within a day would result in start==end)

----- 

Suggestion A:  Truncate start dates and expand end dates to match period type.

Suggestion B: Truncate only start date and leave end date

Suggestion C: Truncate start and end dates 

Suggestion D: Optionally truncate one of A or B

Puckfist, 2018-01-31:
My main belief is,
A user should be able to request data from a start time to an end time of a rollup. No times get adjusted, the final rollup is potentially small.

As far as truncating, I can see arguments for A and B. Is it a distinct option to say expand the final time period, but don't extend it past now?

Really, I can't see myself asking the program to change my input like this (I would just request what I wanted, instead), so I have no intuition on why / when / how to do it.

terrypacker, 2018-02-14:
Currently the REST v2 endpoint can optionally truncate both the start and end times.  I believe that we should remove truncation from Excel Reports all together.

Puckfist, 2018-02-23:
I believe the end date is getting expanded now, not truncated

terrypacker, 2018-03-14:
Need to fix ExcelReportRollupDataPointReference to use the ExpandTimePeriodAdjuster when that is added to the core.

terrypacker, 2018-04-04:
This is done.  Assuming that we expand all dates when doing Rollups we can close this.

terrypacker, 2018-07-16:
Closing as done and released.

closed 2018-07-16 by terrypacker