MangoAutomation/ma-core-public #1222: DataPointRT pointValue initialized in initialize method, count be constructor

Puckfist, 2018-02-23:
For some reason, the pointValue isn't assigned until the initialize() call. The thing that inspired me to move it is that the dprt's created for script validation all have to call resetValue() to get pointValue assigned for a point that isn't actually running right now. This is going to result in duplicated database calls for this, but since it is a user driven action we can stave it off until we resolve if there are other effects of moving its initialization (2x#points database calls when the user clicks validate isn't terrible).

It isn't extra work to assign pointValue in the constructor because the valueCache does its work to get that point value there.

terrypacker, 2018-04-04:
We should review and close or fix for 3.4 release.

Puckfist, 2018-04-25:
We determined calling resetValues is not so bad, and that it was extra work in the constructor if the default cache size for the point was 0 (which would be expanded to 1 when the dprt is initialized). Not changing.

closed 2018-04-25 by Puckfist