MangoAutomation/ma-core-public #1235: DataPointEventMulticaster - unchecked exceptions might cause downstream issues

Puckfist, 2018-03-14:
The DataPointEventMulticaster doesn't catch exceptions from the A leg of point events, which means an unchecked exception bouncing up to that level would result in the B leg not getting executed.

Puckfist, 2018-04-03:
Now the B leg will get executed and the blemish has moved that should both throw runtime exceptions / errors only the B leg's exception will make it out.

Puckfist, 2018-04-09:
We had a discussion about doing something like,

try {
 a.stuff();
} catch(Exception e) {
 b.stuff();
 throw e;
}
b.stuff()

instead of try-finally, with the advantages of not doing more work should a more serious throwable come along than an Exception

terrypacker, 2018-04-12:
Write a JUnit test in the core and then close when that is done.