MangoAutomation/ma-core-public #1236: NPEs in Json emports

Puckfist, 2018-03-21:
The getBoolean / Int / Float / Double / Long methods in JsonObject are quite willing to throw NPEs if the item isn't in the object. A few places don't check that the key exists (and much less that it's the right type) before calling that function. One that I'm not sure what to do about is the "enabled" property of publishers, data sources, data points and abstract action vos. I am about to commit what I think are the obvious fixes...

Puckfist, 2018-03-21:
https://forum.infiniteautomation.com/topic/3365/event-handler-configuration-import-3-2-2/2

Puckfist, 2018-03-21:
https://github.com/infiniteautomation/ma-modules-proprietary/commit/9fb091d624fd741fed2425662a6127201b3e1a6e

Puckfist, 2018-03-21:
In doing this I am finding i think VOs may have intended to use things like the "getBoolean" function in AbstractPointEventDetectorVO which seems totally unused. There would be a lot of extra null checking in the various import methods if that were the case, and currently that method sits unused

Puckfist, 2018-03-21:
With the exception of the enabled property, this is ready for testing now

terrypacker, 2018-03-21:
For 3.4 we should relocate the helper methods i.e. AbstractPointEventDetectorVO .getBoolean() etc into a lower level Abstract VO class so they can be used by all VOs.

terrypacker, 2018-04-03:
The enabled member still needs to be checked.  If it exists use the value if not do nothing.

Puckfist, 2018-04-16:
NPE avoided. Now Advanced Schedules, DataPointVO, DataSourceVO and PublisherVO will permit you not to include an "enabled" value.

Puckfist, 2018-04-23:
There are translation keys specific to event detectors in there.

Puckfist, 2018-04-25:
https://github.com/infiniteautomation/ma-core-public/commit/1f57cec61d0a58dd5fcae5c85048653b6ef75883

terrypacker, 2019-03-20:
Re-opening for Mango 3.6 so we can do another review of the code.

terrypacker, 2019-03-20:
We should consider removing these methods from the Sero Json code which is now part of the core or at least fixing them to not throw ClassCast or NPE's maybe replace them with a 'safe' version.

terrypacker, 2019-03-21:
Note that the current helper methods do not prevent Class Cast Exceptions from leaking out:

When importing a value for a String field that is an int in the JSON using 

` AbstractVO.getString(json, name, key) `

such as

` "updatePeriodType":4, `

would cause a class cast exception.  This would be the same for all helper methods.

terrypacker, 2019-03-21:
I'm considering:

1.  Deprecating the JsonObject.get methods that don't use defaults (or getting rid of them entirely)
2.  Changing the AbstractVO.get... methods to something like this:

```
    protected String getString(JsonObject json, String name) throws JsonException {
        JsonValue o = json.get(name);
        if(o == null)
            throw new TranslatableJsonException(DEFAULT_MISSING_IMPORT_KEY, name);
        try {
            return json.getString(name);
        }catch(ClassCastException e) {
            throw new TranslatableJsonException(DEFAULT_WRONG_DATA_TYPE_KEY, name, String.class.getSimpleName());
        }
    }
```

closed 2019-04-09 by terrypacker