MangoAutomation/ma-core-public #1245: Meta History Endpoint fails

terrypacker, 2018-04-12:
The endpoint does not fully complete its updates but no status is returned about this.  However a log message is printed about a class cast exception in the ScriptPointValueSetter.

terrypacker, 2018-04-12:
```
ERROR 2018-04-11T16:18:28,736 (com.serotonin.m2m2.meta.MetaPointLocatorRT.execute:321) - com.serotonin.m2m2.rt.dataImage.HistoricalDataPoint cannot be cast to com.serotonin.m2m2.rt.dataImage.DataPointRT 
java.lang.ClassCastException: com.serotonin.m2m2.rt.dataImage.HistoricalDataPoint cannot be cast to com.serotonin.m2m2.rt.dataImage.DataPointRT
	at com.serotonin.m2m2.rt.script.ScriptPointValueSetter.set(ScriptPointValueSetter.java:17) ~[classes/:?]
	at com.serotonin.m2m2.rt.script.AbstractPointWrapper.set(AbstractPointWrapper.java:121) ~[classes/:?]
	at com.serotonin.m2m2.rt.script.AbstractPointWrapper.set(AbstractPointWrapper.java:116) ~[classes/:?]
	at jdk.nashorn.internal.scripts.Script$Recompilation$41$30$\^eval\_.__scriptExecutor__(<eval>:1) ~[?:?]
	at jdk.nashorn.internal.scripts.Script$40$\^eval\_.:program(<eval>:5) ~[?:?]
	at jdk.nashorn.internal.runtime.ScriptFunctionData.invoke(ScriptFunctionData.java:637) ~[nashorn.jar:?]
	at jdk.nashorn.internal.runtime.ScriptFunction.invoke(ScriptFunction.java:494) ~[nashorn.jar:?]
	at jdk.nashorn.internal.runtime.ScriptRuntime.apply(ScriptRuntime.java:393) ~[nashorn.jar:?]
	at jdk.nashorn.api.scripting.NashornScriptEngine.evalImpl(NashornScriptEngine.java:449) ~[nashorn.jar:?]
	at jdk.nashorn.api.scripting.NashornScriptEngine.access$200(NashornScriptEngine.java:73) ~[nashorn.jar:?]
	at jdk.nashorn.api.scripting.NashornScriptEngine$3.eval(NashornScriptEngine.java:510) ~[nashorn.jar:?]
	at javax.script.CompiledScript.eval(CompiledScript.java:92) ~[?:1.8.0_162]
	at com.serotonin.m2m2.rt.script.CompiledScriptExecutor.execute(CompiledScriptExecutor.java:89) ~[classes/:?]
	at com.serotonin.m2m2.meta.JavaScriptPointLocatorRT.executeImpl(JavaScriptPointLocatorRT.java:84) ~[meta-ds-3.4.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.meta.MetaPointLocatorRT.execute(MetaPointLocatorRT.java:310) [meta-ds-3.4.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.meta.MetaPointLocatorRT.contextPointUpdated(MetaPointLocatorRT.java:181) [meta-ds-3.4.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.meta.MetaPointLocatorRT.contextPointUpdated(MetaPointLocatorRT.java:166) [meta-ds-3.4.0-SNAPSHOT.jar:?]
	at com.infiniteautomation.mango.rest.v2.MetaPointsGenerationTask.generateMetaPointHistory(MetaPointsGenerationTask.java:263) [meta-ds-3.4.0-SNAPSHOT.jar:?]
	at com.infiniteautomation.mango.rest.v2.MetaPointsGenerationTask.run(MetaPointsGenerationTask.java:156) [meta-ds-3.4.0-SNAPSHOT.jar:?]
	at com.infiniteautomation.mango.rest.v2.temporaryResource.MangoTaskTemporaryResourceManager$1.run(MangoTaskTemporaryResourceManager.java:112) [mango-api-3.4.0-SNAPSHOT.jar:?]
	at com.serotonin.timer.Task.runTask(Task.java:179) [classes/:?]
	at com.serotonin.timer.TaskWrapper.run(TaskWrapper.java:23) [classes/:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_162]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_162]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_162]
```

terrypacker, 2018-04-12:
I think the correct solution for the class cast is to add an isSettable() method to the IDataPointValueSource interface and not do the cast.

terrypacker, 2018-04-12:
The other problem that can be seen in the incomplete node test is that the updates don't seem to match the point values that are inserted within that time range.  This might be an issue with the test though.  I think we should add a cleaner way to return results.  Perhaps a Map<String, PointHistoryGenerationResult> instead of a Map<String,String>.

Puckfist, 2018-04-16:
The issue with the update counts is known: https://github.com/infiniteautomation/ma-core-public/issues/1138

terrypacker, 2018-04-16:
For reference some work has been done on this:
9b962056168154b4110126194270e348380461a7

terrypacker, 2018-04-16:
infiniteautomation/ma-modules-private@d66e72473b9dce6f3381bf36ed6bf637a61f736a
infiniteautomation/ma-modules-private@fbe89e4ca49950ea2fa019d6e538b45a4f1acc2b

closed 2018-04-16 by terrypacker