MangoAutomation/ma-core-public #1248: Create Custom Permissions to Edit Event Handlers Feature Request

terrypacker, 2018-04-13:
If we create Permission Definitions we will need to consider what they apply to, for example currently to edit an event handler you need:

For Data Point Event Handlers:
- Overall DataSource Permission 
- Specific Edit Permission to the owning Data Source

For Publisher,System,Audit Events
- superadmin permission

Joelhaggar, 2018-04-13:
I think we can create a permission definition for the editing and creating of event handlers but only on data points that they have permissions to at least read.  This could also apply to Schedules they have permissions to edit but not system events unless they are super admin.

terrypacker, 2018-04-16:
Not going to do this at this point.

closed 2018-04-16 by terrypacker