MangoAutomation/ma-core-public #1263: All rollup returns invalid JSON with duplicate property names

jazdw, 2018-04-23:
If you try and use the ALL rollup and request the VALUE and RENDERED fields it puts both into the output using the same property names.

Sample query:
`/rest/v2/point-values/time-period/multistate/ALL?fields=TIMESTAMP&fields=VALUE&fields=RENDERED&from=2018-04-23T20:55:51.383Z&limit=5001&timePeriodType=SECONDS&timePeriods=10&timezone=America%2FDenver&to=2018-04-23T21:10:51.383Z&truncate=true`

Sample result:
```
[{"timestamp":1524516950000,"START":1,"FIRST":2,"LAST":3,"COUNT":2,"data":[{"value":1,"starts":0,"runtime":762,"proportion":0.0762},{"value":2,"starts":1,"runtime":5000,"proportion":0.5},{"value":3,"starts":1,"runtime":4238,"proportion":0.4238}],"START":"One","FIRST":"Two","LAST":"Three","COUNT":2,"data":[{"value":"One","starts":0,"runtime":762,"proportion":0.0762},{"value":"Two","starts":1,"runtime":5000,"proportion":0.5},{"value":"Three","starts":1,"runtime":4238,"proportion":0.4238}]}, ...]
```

Puckfist, 2018-09-10:
Fixed here? https://github.com/infiniteautomation/ma-modules-public/commit/fb740165a67607c47b0e59710b6f96ebc3e34222#diff-0a4184171e40931e4837c1ef8bcd32e0

terrypacker, 2018-09-10:
Yes I fixed that, probably didn't know there was a git issue around it as I bumped into it when fixing a different bug.

closed 2018-09-11 by Puckfist