MangoAutomation/ma-core-public #1271: Missing Excel Report Internal Data Point

Joelhaggar, 2018-05-04:
I just noticed that the Excel Report Internal Data Point show how many run reports there are and not the number of report templates there are configured. We should add an Internal Data Point to be created for how many report templates there are configured.

terrypacker, 2018-11-06:
@jazdw there is a monitor that exists for the templates, however it isn't showing up on the Admin UI Home page.

terrypacker, 2018-11-07:
The xid for excel report templates is: `internal_mango_num_report_instances`

terrypacker, 2018-11-07:
As for the reading of 0.0, I can see that many of the points do this which is not correct.  There is code in the InternalMenuItem class that creates these points which will need to be reviewed as it should set the text renderer properly with this code from line 243:

```
						    if(dp.getTextRenderer() instanceof AnalogRenderer && !dp.getXid().equals(SYSTEM_UPTIME_POINT_XID)) {
                                // This are count points, no need for decimals.
                                ((AnalogRenderer)dp.getTextRenderer()).setFormat("0");
                            }
```

jazdw, 2018-11-07:
> As for the reading of 0.0, I can see that many of the points do this which is not correct. There is code in the InternalMenuItem class that creates these points which will need to be reviewed as it should set the text renderer properly with this code from line 243:

Its possible that my data points were created before this code was added. If I delete the points will they be automatically re-created?

> The xid for excel report templates is: internal_mango_num_report_instances

This appears to be the XID for report (not Excel report) instances. Anyway I found that there is already a point for the number of Excel report templates, its XID is `internal_mango_num_excel_report_templates`

jazdw, 2018-11-07:
This commit fixes the XID and links to the new UI for Excel reports.

https://github.com/infiniteautomation/ma-dashboards/commit/eae9f40d5c9885e0508be9ef2173f4b631fbc16a

Leaving open until we sort out the text renderers.

terrypacker, 2019-02-26:
Closing as the renderers are set to whole numbers upon point creation.  Any old points will remain as decimal.

closed 2019-02-26 by terrypacker