MangoAutomation/ma-core-public #1273: watchList vs. watchlist emport

Puckfist, 2018-05-08:
The WatchlistEmportDefinitionNoCase and the WatchListEmportDefinition are both visible on the emport page and both export the entire list of watchlists during backups into their respective keys. This is confusing. I believe these two definitions were to support legacy importing? If so, we should pick one of the keys and only export that one, but still accept importing both?

terrypacker, 2018-05-11:
You are correct they allow importing both 'watchlists' and 'watchLists'.  We should be using the key that was there from the very beginning as for one release a ways back I mistakenly changed the JSON not taking into considering that it was case sensitive.

terrypacker, 2018-07-16:
NOTE that this fix is on the handle-acks branch.

closed 2018-08-20 by Puckfist