MangoAutomation/ma-core-public #1277: Event handling - handle event acknowledgements

Puckfist, 2018-05-16:
Events are thought of not only as active or inactive in Mango, but also acknowledged or unacknowledged. Acknowledging an active event should probably be a handle-able condition, but the handlers don't have it! Perhaps someone may even want to put an acknowledge handler on an inactive event (I believe it's implied an acknowledge handler on an event that doesn't return to normal should fire regardless of event state).

Puckfist, 2018-05-17:
This is pretty close to done in its branch. I think the only outstanding TODO is writing the base acknowledge email template.

Puckfist, 2018-05-17:
https://github.com/infiniteautomation/ma-core-public/commit/1f8d8500522c9df34996a620d25ab4f7b1b6309a

Puckfist, 2018-05-17:
https://github.com/infiniteautomation/ma-core-public/commits/handle-acks

terrypacker, 2018-07-25:
I've found a bug in the logic for Event Handlers by testing this branch.  When an event is acknowledged via the legacy UI (maybe new UI too, not sure).  The EventHandlerRT is set using EventManagerImpl.setHandlers() which creates new RTs.  The ack logic of at least the email event handler requires that the eventRaised() method to have been run (which it won't because this event wasn't in cache and thus has a new RT).  I'll fix this but we need to check the other handlers to see if this type of thing can happen there.

terrypacker, 2018-07-25:
On further investigation this is because the event I was working with a DataPointChange event is not Return applicable and thus is not put into the event manager's active events list.  So basically the ack logic of an event handler cannot depend on a state that is expected to exist because it's RT will be new every time the event is acknowledged.

terrypacker, 2018-07-25:
There is an additional problem for Email handlers that there is no context available on events that are not Rtn applicable so the generation of the template fails.

terrypacker, 2018-07-25:
There is a TODO on line 469 of EventManagerImpl to figure out how to create a context for an acked event.  This is required if we want to provide information on acked events during handling.  We could also enforce the rule that the context is null.