MangoAutomation/ma-core-public #1284: ValueCache can contain AnnotatedPointValueTime and AnnotatedIdPointValueTime objects which cause cast issues

Puckfist, 2018-06-08:
If a cache is > 1, then it could have loaded AnnotatedIdPointValueTime out of the database but get an AnnotatedPointValueTime set into it by the user. This will mean both are in the cache, which will cause code like https://github.com/infiniteautomation/ma-core-public/blob/eeaaa4598b39e3a35ffba66d95313b087b9f6cda/Core/src/com/serotonin/m2m2/web/dwr/DataPointDetailsDwr.java#L84 to break

terrypacker, 2018-06-27:
There are several classes to consider here but the basic solution will be to ensure that the following:
- AnnotatedPointValueTime
- com.infiniteautomation.nosql.maint.IdPointValueTime
- com.serotonin.m2m2.rt.dataImage.IdPointValueTime
- AnnotatedIdPointValueTime

are all castable to PointValueTime

We need to also consider the interfaces:
ITime
IValueTime
IdTime

terrypacker, 2018-06-28:
I think this can cause problems with the v2 REST controller too.  In queries that include realtime data I think a cast exception could happen.

Puckfist, 2018-08-13:
kluged a catch into the old table chart popup: https://github.com/infiniteautomation/ma-core-public/commit/b521810f3c9428973a99cfac7ec70151f3e74826

terrypacker, 2018-08-14:
@Puckfist did you test that code?  I think it will fail when it tries to test sourceMessage for existence on a PointValueTime object.  We may need to create a tag to test for instanceof IAnnotated.

Puckfist, 2018-08-15:
That's why I catch it. It does throw an exception, and yes it was a fix to a bug with that mouseover table popping up

Puckfist, 2018-08-15:
I also can see the original issue is resolved, so I think this can be closed if you're okay with the JSP kluge

closed 2018-08-15 by terrypacker