MangoAutomation/ma-core-public #1289: Upgrade21 fails if duplicate users have done anything

Puckfist, 2018-06-18:
Upgrade 21 tries to be clever and build a list of users to delete into one statement, but it doesn't account for relational data that may have the user as its foreign key, and then the SQL fails. Probably the right thing to do is loop over the IDs and call the UserDao.instance.deleteUser( userId ); method

Puckfist, 2018-06-18:
The risk being that one of the users is probably more active in terms of user comments and whatever, so maybe that'll be the duplicate selected for deletion! Maybe it was better of failing to start?

terrypacker, 2018-06-18:
Or check the relational data to see if there is any.  Delete the user's with no relational data, then if any duplicates are left then maybe fail to start with a useful message or a way to indicate which of the duplicates to delete on next startup.

terrypacker, 2018-07-16:
Closing as the code is in and works.

closed 2018-07-16 by terrypacker