MangoAutomation/ma-core-public #1293: after mango started,then add some serial port which  can't be find on modbus or other  serial  datasource

minghu, 2018-06-26:
after mango started,add serial port like usb-serial convert or  virtual serial software, the serial port can't be found until  restart mango.

terrypacker, 2018-06-26:
Serial ports are only refreshed at startup OR when you click the green refresh button next to the serial port drop down on a data source edit page.  If that doesn't work you may need to modify the serial port regex in the env.properties file.

```
#Regex used to match serial ports so they show up in the menu
serial.port.linux.regex=((cu|ttyS|ttyUSB|ttyACM|ttyAMA|rfcomm|ttyO|COM)[0-9]{1,3}|rs(232|485)-[0-9])
serial.port.linux.path=/dev/
serial.port.windows.regex=
serial.port.windows.path=
serial.port.osx.path=/dev/
serial.port.osx.regex=cu.USB0
```

Puckfist, 2018-08-16:
And if a data source is enabled to use a serial port identifier that is not seemingly enumerated in the dropdown list, it will attempt to open the described serial port as a JSSC serial port with the settings configured on the data source

Puckfist, 2018-09-10:
Closing as insufficient information to reproduce issue / explained as expected behavior

closed 2018-09-10 by Puckfist