MangoAutomation/ma-core-public #1294: JFree charts default font language support

Puckfist, 2018-07-03:
It was observed in a support email that the JFree charts that include legends do not render certain characters, like Chinese, properly. This is caused by the default font not supporting those characters. I was able to render Chinese characters upon adding the following to com.serotonin.m2m2.util.chart.ImageChartUtils in writeChart

Font font = new Font("Open Sans, Lucida Sans", Font.PLAIN, 12);
LegendTitle lt = chart.getLegend();
if(lt != null)
    lt.setItemFont(font);

and also,

anno.setFont(font);

Not sure if we want to use a system setting to set a preferred font or change the default to one with more language support?

terrypacker, 2018-08-23:
I'm open to making it an env/system setting probably would be pretty easy.

terrypacker, 2018-08-23:
If its a system setting you probably need an endpoint to return a list of available fonts to put into the drop down.

closed 2018-08-29 by Puckfist