MangoAutomation/ma-core-public #1296: Scripts in CSV exports not character escaped

Puckfist, 2018-07-05:
There is no character escaping in CSV exports of meta point scripts, which leads to files you cannot import back in. I suspect the solution will involve encoding the newline, comma, and double quote characters. Perhaps something like....

```
	@CSVColumnGetter(order=18, header="script")
	public String getScriptCsv() {
	    return this.data.getScript().replaceAll("\\n", "\\\\n").replaceAll(",", "~&").replaceAll("\"", "\\\\\""); //~& would not be valid code
	}
	
	@CSVColumnSetter(order=18, header="script")
	public void setScriptCsv(String script) {
	    this.data.setScript(script.replaceAll("\\\\\"", "\"").replaceAll("~&", ",").replaceAll("\\\\n", "\n"));
	}

```
Upon testing I found the Bulk Data Point Editor still had the original script exporting, probably because it is done via some relfection mechanism and not the CSV annotations? This will be a problem whenever another something with a script is exported to CSV, so we may need some util CSV encode / decode someplace

terrypacker, 2018-07-05:
I was just able to successfully export this script:

```
//Test comment
log('this is a test');
log("this is a test");
log("this, is , another test");
if(p13.value > 10){
    return true;
}else{
    return false;
}
```

and it exported like this:
```
,"
//Test comment
log('this is a test');
log(""this is a test"");
log(""this, is , another test"");
if(p13.value > 10){
    return true;
}else{
    return false;
}"
```

What am I missing?

jazdw, 2018-07-05:
I can download and edit a CSV file with meta points in LibreOffice, they re-import fine. The key is that the script field must be double quoted, it shouldn't matter that it has a line break in it.

It works via Jackson @Puckfist , the code is in GenericCSVMessageConverter and yeah we could make a more generic class to handle encoding/decoding elsewhere. The @CSVColumnGetter are for the old CSV converters and are not used.

We can escape the line breaks if we really need to.

Puckfist, 2018-07-05:
My mistake: I didn't even try reimporting as I figured it's not a CSV file with line breaks and commas like that, but it still imported fine!. I suspect the issue with Excel is that you need to specify the delimiter when importing a CSV into excel, and there may be no setting for this situation.

closed 2018-07-05 by Puckfist