MangoAutomation/ma-core-public #1298: Z-Wave Java 9 Compatibility

terrypacker, 2018-07-16:
Running java 9 there may be a problem with the Z-Wave data source:

Exception in thread "Thread-1" java.lang.NoClassDefFoundError: javax/xml/bind/JAXBException
	at com.infiniteautomation.zwave.ZwaveDataSourceDefinition.postTerminate(ZwaveDataSourceDefinition.java:71)
	at com.serotonin.m2m2.module.Module.postTerminate(Module.java:179)
	at com.serotonin.m2m2.Lifecycle.terminate(Lifecycle.java:485)
	at com.serotonin.m2m2.Main$1.run(Main.java:129)
Caused by: java.lang.ClassNotFoundException: javax.xml.bind.JAXBException
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:466)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:563)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:496)

terrypacker, 2019-05-10:
I tested both Java 8 and Java 12 and found that its fixed by adding the javax/xml dependencies to the core.

terrypacker, 2019-05-17:
Added javax/xml in this commit a7f1d8ee596855f09ea92c91b67aee3ff8273976

terrypacker, 2019-11-04:
I think we also need to add in a dependency for `javax.activation` as this is used in our `com.serotonin.web.mail.EmailAttachment class`.

terrypacker, 2019-11-04:
```
<dependency>
    <groupId>com.sun.activation</groupId>
    <artifactId>javax.activation</artifactId>
    <version>1.2.0</version>
</dependency>
```

jazdw, 2019-11-05:
@terrypacker Please confirm if the Java 9 compatibility extension script is required.

terrypacker, 2019-11-20:
We don't actually use the`javax.activation` code in our Email logic and I tested sending attachments in Java 9 without the script, all works fine.  Closing this now.

closed 2019-11-20 by terrypacker