MangoAutomation/ma-core-public #1299: Log file and backup accumulation

jazdw, 2018-07-17:
Just logged into a client's server and tried to get to the log files, they had over 5000 log files. This is silly, no one is ever going to read that many log files.

File names seem to mainly consist of things like `metaPoint-7448.log`, `tsdb-series-1752-shard-700-scan-report_2017-09-05_18-15-14.log` etc.

Proposal

1. Only store useful information, such as errors, if its all fine don't log anything
2. Store all information in a single file, e.g. not per meta point or per series/shard
3. Rotate logs, keeping only the latest 10 log files or something like that

The same issue applies to backups. They are not being properly rotated/deleted. Another customer's server has crashed several times as its run out of disk space to due to the "incremental" TSDB backups.

terrypacker, 2018-07-17:
The Meta logs could use the RollingProcessLog or RollingIOLog framework as should all other references to IOLog or ProcessLog.

terrypacker, 2018-11-06:
I would consider this issue done as of 3.5 but I'm leaving it open for 3.6 just in case.

jazdw, 2018-11-07:
@terrypacker the backups are being rotated now?

terrypacker, 2018-11-07:
@jazdw no I was focused on the logging aspect of this and didn't think about the backups.  There is no rotation for backups implemented yet.

terrypacker, 2018-11-07:
Oh sorry, yes 
1.  the NoSQL does incremental and allows you to keep x number of files.
2.  The JSON Configuration allows keeping x number of files
3.  The SQL Backup allows keeping x number of files 

What else is there?

jazdw, 2018-11-07:
What are the defaults for x number of files? Was this a new feature that was implemented recently?

terrypacker, 2018-11-07:
They are system settings that have been here for awhile.

1. NoSQL Defaults to 365 incremental Daily Backups
2. JSON Config Defaults to 10 Daily Backups
3. SQL Backup Defaults to 10 Daily Backups

jazdw, 2018-11-07:
> NoSQL Defaults to 365 incremental Daily Backups

OK I see this as an issue, as the backups are not truly incremental. They are massive!

closed 2020-12-04 by terrypacker