MangoAutomation/ma-core-public #1301: Polling Data Source - Aborted Polls Cause Blocking

terrypacker, 2018-07-20:
This commit: a4a0276e49ad2ab4a3a80fc7cd2c245ac71c33e0 added synchronization to the poll method which prevents us from properly tracking aborted polls and in effect blocks the next scheduled poll.

Review to ensure this is the best way sync the force poll and scheduled poll interaction.  Some ideas:

1.  If we keep synchronization we should override the TimeoutClient to have a queue size of 0 so that any polls that try to run cannot if the first one is running.

2.  Another option is to check the jobThread field in the forcePoll() method and wait until it is free to perform the force poll.  Some form of wait() notify() logic would be acceptable.

Puckfist, 2018-07-23:
I think the only way to have more than one thread trying to get into that method at the same time would be if someone was forcing polling. Shouldn't the aborted poll counter be incremented by the task queue rejecting the second scheduled execution of the polling data source at line 98 of PollingDataSource ?

Puckfist, 2018-07-23:
Ah, I was thinking (1) already applied, but I see in the OrderedThreadPoolExecutor:211 that the first task would not appear in the queue, and the default queue size is 1, so you are correct but even still shouldn't the enqueued task not be executed until the thread in the poll returns (at which point it will execute the next task)?

terrypacker, 2018-07-25:
I've taken another look at this and I think the main goal here is to avoid the overhead from the newly added synchronized keyword since it is only required during a Force Poll scenario.  I've come up with a few ways to do this but none guarantee that both the forced poll will run AND synchronization.  Here is my best idea:

In the forcePoll method we could submit a timeout task to the time which will potentially insert it into the queue of the polling task. If a poll is currently running it will run next or be added to the aborted polls list, if no poll is running it will be run straight away.

Puckfist, 2018-08-01:
Interesting suggestion! I don't have any issue with that, but I could see an argument being that would mean an asynchronous poll to forcing the poll, so one could not assure oneself in a script that the line after a forcePoll would have the results of that poll in it. Similarly, one could not implement the "500 ms between polls" in a script, since refreshDataSource would fall right through. One would have to cook up that logic using event handlers or meta points.

I wonder, is the only downside to the queue size of 0 that the synchronization overhead exists?

closed 2018-08-14 by terrypacker