MangoAutomation/ma-core-public #1303: Data Point Tags - Odd behavior

terrypacker, 2018-07-21:
I have noticed that:

1.  The name and device tags are not in the DataPointRT.vo.tags list. (This is getting set before scripting, when a point is updated, before email template generation and realtime data queries)

2.  I have name and device entries in my tags database table.

jazdw, 2018-07-23:
This is by design. The name and device tags can be considered pseudo tags. They are updated by setting the name and deviceName properties of the data point itself.

Puckfist, 2018-07-23:
This is why in several places for the tags you see it set into the DataPointVO and then the tag map returned from the function is the tag map returned from the DataPointVO's getTags

terrypacker, 2018-07-23:
We should probably define the logic and place it into a common area for loading/saving the tags for a data point.

Currently the code I wrote won't have the name or device tags loaded into the map which I think is wrong.

terrypacker, 2018-07-23:
Closing as fixed.  This is the logic to use when getting tags for a DataPointVO that has tags==null
```
DataPointVO vo = DataPointDao.instance.get(dataPointId);
            Map<String, String> tags = DataPointTagsDao.instance.getTagsForDataPointId(dataPointId);
            vo.setTags(tags);
            return vo.getTags();
```

closed 2018-07-23 by terrypacker