MangoAutomation/ma-core-public #1304: Mango Javascript PointWrapper Statistics Don't Match REST api

terrypacker, 2018-08-01:
The v2 REST api uses a Bookending algorithm to set the start value of the StatisticsGenerator classes.  This is different to how the Javascript API works.

The javascript api gets the value before the period and then the values within the period.  This is different to Bookending as if there is a value exactly at the start of the period the Javascript api will not use that as the startValue.

The ideal solution will use the Bookending query to perform this, however the Javascript point uses the DataPointRT to get its values which is cache aware.  We should consider implementing a bookending style method to the RT (as like in the PointValueFacade)

terrypacker, 2018-08-01:
It looks like the ExcelStatisticsGenerator has the same problem.

Puckfist, 2018-08-01:
I think they should be consistent, good catch. Easy fix, change getPointValueBefore to from+1, between with from+1, add value to beginning of values list if at time from, done

There is a way to get samples from the bookending endpoints, I believe the PointValueTimeStreamScriptUtility provides those options

terrypacker, 2018-08-01:
I've changed Excel Reports to the a bookend query for statistics.  See commit infiniteautomation/ma-modules-proprietary@380bcb9ca7288ca2fe1b7591a294f183dee56a5a

terrypacker, 2018-08-01:
Excel reports does not have the option to use cached values.

terrypacker, 2018-08-01:
The javascript stats have been fixed by @Puckfist  suggestion to alter query times.  A longer term solution that would allow optionally using cached values is discussed here: #1305

closed 2018-08-22 by Puckfist