MangoAutomation/ma-core-public #1305: Optionally Use Cache in Javascript Statistics

terrypacker, 2018-08-01:
All javascript statistics currently use a DataPointRT to query the values, which defaults to using the cache.  Ideally one could choose to use the cached values or not when making a request for statistics.

The solution will be to implement a class based on the logic in the PointValueFacade but implementing the IDataPointValueSource interface.

Be sure to include the cache processing logic from the PointValueFacade.

Puckfist, 2018-08-15:
Just tossing a note in here that there was a discussion of how lastValue() would be not using the cache, but because there is a delay between handing the value to the database controller and it actually being written, a context point that logs all data could generate an execution event, but it's lastValue() would not be it's .value

If one puts in a delay on execution, then they are equal. The value is still accessible through either .lastValue(true) to use the cache, or through the .value property.

Perhaps more poignant would be an interval logging point and a meta point that was triggered by logged events. The value of the logging is probably not available through either lastValue method if there's an execution delay of 0 (it wouldn't have actually be written into the database by the time the meta point executes)

Puckfist, 2019-01-02:
Re-opening as the lastValue and last functions have issues when not using cache in history generation, because the valueFacade in the AbstractPointWrapper doesn't have a concept of being in a history generation state, and so it requests current last values.

closed 2019-07-10 by Puckfist