MangoAutomation/ma-core-public #1310: Log4J Data Source Infinite Loop

terrypacker, 2018-08-03:
If a LOG message gets generated while processing the Log4jDataSourceAppender's WorkItem it will cause another WorkItem to get created in a loop.

Same issue applies to the Slack Publisher.

Puckfist, 2018-08-03:
Did you observe this somehow? I agree that it sounds technically possible, but the only LOG message I see  it possibly calling is that for future dated point value rejection, which shouldn't be possible with the time coming from the logged event (unless the timer is in simulation, or something). I don't see a pathway in either the SQL or NoSQL databases where the log4j data source would produce another log message without it being from a runtime exception (like full disk, full memory, whatever)

I think it's doomed to that problem, though. I guess one could decouple saving the values from the appender's thread (not sure what you meant WorkItem, I don't see any WorkItem references in the Log4J Data Source project), but that wouldn't prevent the infinite log dump you describe. I guess maybe if the new logged event were at the same millisecond as the point's current value and way the same message one could abort trying to set it again, but how does one detect the case where the log message is generated but nothing is updated?

I guess to mitigate the uncaught runtime exception possibility one could wrap its update point value calls in some logging alternative try-catch, like raising an event maybe, //pseudo raiseEvent(AlarmLevels.URGENT, "Log event from processing log events averted: " + e.getMessage())

Puckfist, 2018-08-03:
^ Hadn't pulled :(

Puckfist, 2018-08-03:
^ Has the simple solution that if the same messages occur rapidly, those within the message repeat interval will not be handled (unless another type of log message happens in the mean time)

Puckfist, 2018-08-03:
I don't think this issue needs to be resolved in the pathological case (i.e. someone updates a script by the log4j point, then logs something from the script that is different, etc.)

closed 2018-11-06 by terrypacker