MangoAutomation/ma-core-public #1312: RealTimeDataPointValueCache.getUserView() does not honor data source edit permission

jazdw, 2018-08-06:
If you have edit permission to a data source but not read/set permissions to a data point you will not see it in the list of points returned from RealTimeDataPointValueCache.getUserView().

RealTimeDataPointValue should be updated so it can return the DataPointVO and then we can just use Permissions.hasDataPointReadPermission().

There is also another issue in that if RealTimeDataPointValue.initialize() is called the properties such as readPermission etc are not updated. All the getters should be updated to just get the value from the VO instead of storing them inside RealTimeDataPointValue.

terrypacker, 2018-11-06:
Some notes:

1.  The path member of the RealTimeDataPointValue is ok because the parent cache is reset if the hierarchy changes.
2.  Review the getTags() method
3.

terrypacker, 2019-02-28:
The permissions issue is fixed as the cascading call starting at this line in `RealTimeDataPointValueCache.getUserView(User user)`

```
        //Filter on permissions
        for (RealTimeDataPointValue rtdpv : this.realTimeData) {
            if(Permissions.hasDataPointReadPermission(user, rtdpv.vo))
                results.add(rtdpv);
        }
```

Does end up checking for both the Global Data Source and specific Data Source edit permissions.

I believe the `RealTimeDataPointValue.pointInitialized()` part of this issue is now fixed by replacing the VO when the point is initialized and then loading the tags on demand when called.

closed 2019-04-09 by terrypacker