MangoAutomation/ma-core-public #1313: Lifecycle.terminate() order

jazdw, 2018-08-13:
I got this NPE while shutting down Mango.
```
Exception in thread "Mango Restart Task" java.lang.NullPointerException
    at com.serotonin.m2m2.rt.EventManagerImpl.returnToNormal(EventManagerImpl.java:332)
    at com.serotonin.m2m2.rt.EventManagerImpl.returnToNormal(EventManagerImpl.java:291)
    at com.serotonin.m2m2.rt.event.type.SystemEventType.returnToNormal(SystemEventType.java:111)
    at com.serotonin.m2m2.TimedLicenseRegistrar.shutdown(TimedLicenseRegistrar.java:137)
    at com.serotonin.m2m2.Lifecycle.terminate(Lifecycle.java:489)
    at com.serotonin.m2m2.Lifecycle$5.run(Lifecycle.java:1509)
```

`Providers.get(ITimedLicenseRegistrar.class).shutdown();` tries to use `Common.backgroundProcessing` directly after it has been set to null in Lifecycle.terminate(). I fixed the NPE but someone should look at the order these things are shutdown.

We should also wrap each section of terminate() in a try/catch block to ensure that terminate() does not exit early. In particular if there are any exceptions early in the terminate() method then the MangoNoSql module's postTerminate() hook will not be executed and there will be a couple of IasTsdb maintenance threads left hanging around preventing Java from exiting.

terrypacker, 2018-08-14:
I added try/catch clauses around all the sub component shutdown methods.

terrypacker, 2018-08-14:
What you are seeing is due to the new UserEventNotify infrastructure.  We will need to review this as the EventManager requires BackgroundProcessing to be operational to notify Users of event state changes.  I can definitely see the order of things is not right for shutdown.