MangoAutomation/ma-core-public #1326: No way to remove a template from a Data Point via bulk editor or CSV import

Joelhaggar, 2018-09-10:
I have a bunch of data points that were created with the default templates and now I want to edit their properties and not use a Data Point Template.  If I import a CSV with the templateXID black it does not remove the template from the data point.  If I try to update points with a blank template in the bulk editor I get errors.

jazdw, 2018-10-16:
@Joelhaggar with the CSV if you leave a cell blank it basically means "no change", the field is undefined. There are several key word strings which you can enter, among them NULL and EMPTY. If you set the cell to NULL then it will remove the template XID.

You're right it can't be done in the bulk editor. I'll fix that (put a delete button above the column like tags).

jazdw, 2018-10-16:
Fixed in https://github.com/infiniteautomation/ma-dashboards/commit/936cc4d1586e0e599ec9aac4401429809092a3cd

closed 2018-10-16 by jazdw