MangoAutomation/ma-core-public #1328: Compile ma-core-public/CoreBundles failed

aploese, 2018-09-13:
this is the last output from mvn -X install, the same happens in ma-modules-public.

>  [DEBUG] Could not find metadata com.infiniteautomation.mango:mango-api:3.5.0-SNAPSHOT/maven-metadata.xml in local (/home/aploese/.m2/repository)
[DEBUG] Using transporter WagonTransporter with priority -1.0 for https://maven.mangoautomation.net/repository/ias-snapshot/
[DEBUG] Using connector BasicRepositoryConnector with priority 0.0 for https://maven.mangoautomation.net/repository/ias-snapshot/
Downloading from ias-snapshots: https://maven.mangoautomation.net/repository/ias-snapshot/com/infiniteautomation/mango/mango-api/3.5.0-SNAPSHOT/maven-metadata.xml
Downloaded from ias-snapshots: https://maven.mangoautomation.net/repository/ias-snapshot/com/infiniteautomation/mango/mango-api/3.5.0-SNAPSHOT/maven-metadata.xml (968 B at 2.4 kB/s)
[DEBUG] Writing tracking file /home/aploese/.m2/repository/com/infiniteautomation/mango/mango-api/3.5.0-SNAPSHOT/resolver-status.properties
[DEBUG] Could not find metadata com.infiniteautomation.mango:mango-api:3.5.0-SNAPSHOT/maven-metadata.xml in local (/home/aploese/.m2/repository)
[DEBUG] Skipped remote request for com.infiniteautomation.mango:mango-api:3.5.0-SNAPSHOT/maven-metadata.xml, already updated during this session.
[DEBUG] Could not find metadata com.infiniteautomation.mango:mbus-ds:3.5.0-SNAPSHOT/maven-metadata.xml in local (/home/aploese/.m2/repository)
[DEBUG] Using transporter WagonTransporter with priority -1.0 for https://maven.mangoautomation.net/repository/ias-snapshot/
[DEBUG] Using connector BasicRepositoryConnector with priority 0.0 for https://maven.mangoautomation.net/repository/ias-snapshot/
Downloading from ias-snapshots: https://maven.mangoautomation.net/repository/ias-snapshot/com/infiniteautomation/mango/mbus-ds/3.5.0-SNAPSHOT/maven-metadata.xml
Downloaded from ias-snapshots: https://maven.mangoautomation.net/repository/ias-snapshot/com/infiniteautomation/mango/mbus-ds/3.5.0-SNAPSHOT/maven-metadata.xml (966 B at 2.4 kB/s)
[DEBUG] Writing tracking file /home/aploese/.m2/repository/com/infiniteautomation/mango/mbus-ds/3.5.0-SNAPSHOT/resolver-status.properties
[DEBUG] Could not find metadata com.infiniteautomation.mango:mbus-ds:3.5.0-SNAPSHOT/maven-metadata.xml in local (/home/aploese/.m2/repository)
[DEBUG] Skipped remote request for com.infiniteautomation.mango:mbus-ds:3.5.0-SNAPSHOT/maven-metadata.xml, already updated during this session.
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 9.658 s
[INFO] Finished at: 2018-09-13T10:52:12+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] 10603
java.lang.ArrayIndexOutOfBoundsException: 10603
    at org.codehaus.plexus.util.xml.pull.MXParser.parsePI (MXParser.java:2502)
    at org.codehaus.plexus.util.xml.pull.MXParser.parseEpilog (MXParser.java:1604)
    at org.codehaus.plexus.util.xml.pull.MXParser.nextImpl (MXParser.java:1434)
    at org.codehaus.plexus.util.xml.pull.MXParser.next (MXParser.java:1131)
    at org.apache.maven.model.io.xpp3.MavenXpp3Reader.read (MavenXpp3Reader.java:4133)
    at org.apache.maven.model.io.xpp3.MavenXpp3Reader.read (MavenXpp3Reader.java:872)
    at org.apache.maven.model.io.DefaultModelReader.read (DefaultModelReader.java:109)
    at org.apache.maven.model.io.DefaultModelReader.read (DefaultModelReader.java:82)
    at org.apache.maven.model.building.DefaultModelProcessor.read (DefaultModelProcessor.java:84)
    at org.apache.maven.model.building.DefaultModelBuilder.readModel (DefaultModelBuilder.java:536)
    at org.apache.maven.model.building.DefaultModelBuilder.readParentExternally (DefaultModelBuilder.java:1098)
    at org.apache.maven.model.building.DefaultModelBuilder.readParent (DefaultModelBuilder.java:830)
    at org.apache.maven.model.building.DefaultModelBuilder.build (DefaultModelBuilder.java:332)
    at org.apache.maven.repository.internal.DefaultArtifactDescriptorReader.loadPom (DefaultArtifactDescriptorReader.java:314)
    at org.apache.maven.repository.internal.DefaultArtifactDescriptorReader.readArtifactDescriptor (DefaultArtifactDescriptorReader.java:192)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.resolveCachedArtifactDescriptor (DefaultDependencyCollector.java:539)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.getArtifactDescriptorResult (DefaultDependencyCollector.java:524)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.processDependency (DefaultDependencyCollector.java:412)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.processDependency (DefaultDependencyCollector.java:365)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.process (DefaultDependencyCollector.java:353)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.doRecurse (DefaultDependencyCollector.java:507)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.processDependency (DefaultDependencyCollector.java:460)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.processDependency (DefaultDependencyCollector.java:365)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.process (DefaultDependencyCollector.java:353)
    at org.eclipse.aether.internal.impl.DefaultDependencyCollector.collectDependencies (DefaultDependencyCollector.java:256)
    at org.eclipse.aether.internal.impl.DefaultRepositorySystem.collectDependencies (DefaultRepositorySystem.java:282)
    at org.apache.maven.project.DefaultProjectDependenciesResolver.resolve (DefaultProjectDependenciesResolver.java:169)
    at org.apache.maven.lifecycle.internal.LifecycleDependencyResolver.getDependencies (LifecycleDependencyResolver.java:243)
    at org.apache.maven.lifecycle.internal.LifecycleDependencyResolver.resolveProjectDependencies (LifecycleDependencyResolver.java:147)
    at org.apache.maven.lifecycle.internal.MojoExecutor.ensureDependenciesAreResolved (MojoExecutor.java:246)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:200)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:154)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:146)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:956)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:290)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:194)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:564)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:289)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:229)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:415)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:356)
[ERROR]

terrypacker, 2018-09-25:
I would suggest cleaning your .m2 directory as I don't see this when I compile via mvn install in the ma-core-public/CoreBundle directory.

terrypacker, 2018-09-25:
Note that you need access to the internet to download all the modules/dependencies from maven.mangoautomation.net

closed 2019-04-09 by terrypacker