MangoAutomation/ma-core-public #1331: Unit Aliases display as Derived Units

terrypacker, 2018-09-27:
As per this thread: 

[https://forum.infiniteautomation.com/topic/3797/mango-changes-mvar-units-to-mva](https://forum.infiniteautomation.com/topic/3797/mango-changes-mvar-units-to-mva)
I would expect the UnitUtil class to allow inputting VAR or VA and then having the returned unit displayed as VAR or VA.  Instead the label isn't getting used and the result is the V*A is displayed as the unit totally ignoring the alias label.

Puckfist, 2019-07-10:
I believe there is a way to prefer one or another alias, but that would have the issue of now V*A would appear as VAR

snowwindwaverider, 2019-07-19:
i am an electical power generation and distribution engineer.  Nobody writes VA as V*A. Units you want to be able to display are A, V, VA, W, VAR, Wh, VARh, VAh, and with all associated SI prefixes mu,m,k,M,G,T. So not being able to display V*A as units is of no significance since it is not a correct way to write any electrical units.

jazdw, 2019-07-19:
You appear to contradict yourself. Can you clarify?
> Nobody writes VA as VA.

> Units you want to be able to display are A, V, VA, W, VAR, Wh, VARh, VAh

I think this issue (https://github.com/infiniteautomation/ma-core-public/issues/309) might give you some more clarity, and honestly this is all the same discussions so I am going to close this one.

closed 2019-07-19 by jazdw

snowwindwaverider, 2019-07-20:
there should have been an asterisk in there