MangoAutomation/ma-core-public #1335: Mango Javascript - Global Scripts out of Sync

terrypacker, 2018-10-03:
It was found that if one updates a global script (or adds one) it is not available to currently running Meta and Scripting data points/sources.  We need to ensure that the global scripts are updated for every run.

terrypacker, 2018-10-04:
We are not updating the scripts on every run, this is specifically warned about in the help.  However we will fix the problem at Import time by adding some code to the Importer so that Global Scripts are imported first.

Puckfist, 2018-10-09:
I'm not sure about doing this in 3.5.1

I would think the right way would be giving a priority method to EmportDefintiion and then do the loop at ImportTask::261 before handling the importers list and only importing ImportItems that have the first priority. But maybe there is a simpler way....?

terrypacker, 2018-10-10:
Adding a priority is the correct way to do that and should be done in Mango 3.6.0, until the I have implemented a solution that hard codes the JSON field name into the ImportTask.  

For 3.6 we should also consider adding a boolean to the EmportDefinition that will add the job as an ImportItem or Importer if it has no dependencies.

Puckfist, 2018-10-10:
Seems to work. Not closing until permanent solution?

terrypacker, 2018-10-10:
Agreed, moving to 3.6.

terrypacker, 2019-05-10:
Calling this good.  The MangoJavaScriptService ensures that the scripts are added every run of a script.

closed 2019-05-10 by terrypacker