MangoAutomation/ma-core-public #1337: Events Websocket - Not publishing events properly

terrypacker, 2018-10-03:
The events web socket seems to have an issue that prevents all events from being sent out and also from clients being able to successfully terminate their sessions.

jazdw, 2018-10-03:
Here's a debug log, with annotations inline.

```
Open 3 tabs each which subscribe to events via websocket
DEBUG 2018-10-03T16:04:24,077 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler.afterConnectionEstablished:46) - Connection established, session ID 326bcbb2 
DEBUG 2018-10-03T16:04:24,417 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler.handleTextMessage:76) - Received message from session 326bcbb2, {"eventTypes":["RAISED","ACKNOWLEDGED","RETURN_TO_NORMAL","DEACTIVATED"],"levels":["LIFE_SAFETY","CRITICAL","URGENT","WARNING","IMPORTANT","INFORMATION","NONE"]} 
DEBUG 2018-10-03T16:04:35,952 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler.afterConnectionEstablished:46) - Connection established, session ID 30a8aec7 
DEBUG 2018-10-03T16:04:37,417 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler.handleTextMessage:76) - Received message from session 30a8aec7, {"eventTypes":["RAISED","ACKNOWLEDGED","RETURN_TO_NORMAL","DEACTIVATED"],"levels":["LIFE_SAFETY","CRITICAL","URGENT","WARNING","IMPORTANT","INFORMATION","NONE"]} 
DEBUG 2018-10-03T16:04:43,541 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler.afterConnectionEstablished:46) - Connection established, session ID 3d7c6266 
DEBUG 2018-10-03T16:04:44,799 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler.handleTextMessage:76) - Received message from session 3d7c6266, {"eventTypes":["RAISED","ACKNOWLEDGED","RETURN_TO_NORMAL","DEACTIVATED"],"levels":["LIFE_SAFETY","CRITICAL","URGENT","WARNING","IMPORTANT","INFORMATION","NONE"]} 

Raise an event, all 3 websockets are notified successfully
DEBUG 2018-10-03T16:05:06,525 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler$EventsWebSocketListener.raised:180) - Event raised, notifying session 3d7c6266: EventInstance [id=23727, eventType=DataPointEventType(dataPointId=1010, detectorId=545), activeTimestamp=1538604306464, rtnApplicable=true, rtnTimestamp=0, rtnCause=0, alarmLevel=4, message=com.serotonin.m2m2.i18n.TranslatableMessage@6c9de4b, eventComments=[], handlers=[com.serotonin.m2m2.rt.event.handlers.EmailHandlerRT@5a44816d], acknowledgedTimestamp=0, acknowledgedByUserId=0, acknowledgedByUsername=null, alternateAckSource=null, hasComments=false, userNotified=false, silenced=false, idsToNotify=[1], context={pointEventDetector=id: 545 name: , point=DataPointVO [id=1010, xid=DP_d29044ef-3c1d-44e3-9f3c-49c401f1a845, name=Slave 1 monitor, dataSourceId=226, deviceName=Vsite005, enabled=true, pointFolderId=0, loggingType=1, intervalLoggingPeriodType=2, intervalLoggingPeriod=15, intervalLoggingType=1, tolerance=0.0, purgeOverride=false, purgeType=7, purgePeriod=1, textRenderer=com.serotonin.m2m2.view.text.PlainRenderer@54836455, chartRenderer=com.serotonin.m2m2.view.chart.TableChartRenderer@50bcea1d, eventDetectors=[id: 545 name: ], comments=null, defaultCacheSize=1, discardExtremeValues=false, discardLowLimit=-1.7976931348623157E308, discardHighLimit=1.7976931348623157E308, unit=, integralUnit=s, renderedUnit=, useIntegralUnit=false, useRenderedUnit=false, rollup=NONE, plotType=1, pointLocator=com.serotonin.m2m2.virtual.vo.VirtualPointLocatorVO@15be6e6e, dataSourceTypeName=VIRTUAL, dataSourceName=DS_Vsite001, dataSourceXid=DS_Vsite001, lastValue=null, overrideIntervalLoggingSamples=false, intervalLoggingSampleWindowSize=0, readPermission=, setPermission=, templateId=null, preventSetExtremeValues=false, setExtremeLowLimit=-1.7976931348623157E308, setExtremeHighLimit=1.7976931348623157E308]}] 
DEBUG 2018-10-03T16:05:06,529 (com.serotonin.m2m2.web.mvc.rest.v1.websockets.events.EventsWebSocketHandler$EventsWebSocketListener.raised:180) - Event raised, notifying session 30a8aec7: EventInstance [id=23727, eventType=DataPointEventType(dataPointId=1010, detectorId=545), activeTimestamp=1538604306464, rtnApplicable=true, rtnTimestamp=0, rtnCause=0, alarmLevel=4, message=com.serotonin.m2m2.i18n.TranslatableMessage@6c9de4b, eventComments=[], handlers=[com.serotonin.m2m2.rt.event.handlers.EmailHandlerRT@5a44816d], acknowledgedTimestamp=0, acknowledgedByUserId=0, acknowledgedByUsername=null, alternateAckSource=null, hasComments=false, userNotified=false, silenced=false, idsToNotify=[1], context={pointEventDetector=id: 545 name: , point=DataPointVO [id=1010, xid=DP_d29044ef-3c1d-44e3-9f3c-49c401f1a845, name=Slave 1 monitor, dataSourceId=226, deviceName=Vsite005, enabled=true, pointFolderId=0, loggingType=1, intervalLoggingPeriodType=2, intervalLoggingPeriod=15, intervalLoggingType=1, tolerance=0.0, purgeOverride=false, purgeType=7, purgePeriod=1, textRenderer=com.serotonin.m2m2.view.text.PlainRenderer@54836455, chartRenderer=com.serotonin.m2m2.view.chart.TableChartRenderer@50bcea1d, eventDetectors=[id: 545 name: ], comments=null, defaultCacheSize=1, discardExtremeValues=false, discardLowLimit=-1.7976931348623157E308, discardHighLimit=1.7976931348623157E308, unit=, integralUnit=s, renderedUnit=, useIntegralUnit=false, useRenderedUnit=false, rollup=NONE, plotType=1, pointLocator=com.serotonin.m2m2.virtual.vo.VirtualPointLocatorVO@15be6e6e, dataSourceTypeName=VIRTUAL, dataSourceName=DS_Vsite001, dataSourceXid=DS_Vsite001, lastValue=null, overrideIntervalLoggingSamples=false, intervalLoggingSampleWindowSize=0, readPermission=, setPermission=, templateId=null, preventSetExtremeValues=false, setExtremeLowLimit=-1.7976931348623157E308, setExtrem [... cut ...] Unit=, useIntegralUnit=false, useRenderedUnit=false, rollup=NONE, plotType=1, pointLocator=com.serotonin.m2m2.virtual.vo.VirtualPointLocatorVO@15be6e6e, dataSourceTypeName=VIRTUAL, dataSourceName=DS_Vsite001, dataSourceXid=DS_Vsite001, lastValue=null, overrideIntervalLoggingSamples=false, intervalLoggingSampleWindowSize=0, readPermission=, setPermission=, templateId=null, preventSetExtremeValues=false, setExtremeLowLimit=-1.7976931348623157E308, setExtremeHighLimit=1.7976931348623157E308]}] 
```

jazdw, 2018-10-04:
Confirmed fixed for me. Thanks Terry.

closed 2018-10-29 by terrypacker