MangoAutomation/ma-core-public #1338: DataPointEventTypeModel - Duplicate Handling Buggy

terrypacker, 2018-10-04:
I noticed that the model property `duplicateHandling` is returned as a String but set via an int.  I'm a bit hesitant to change this if it is being used in the new UI.  It is an easy change but may cause downstream problems.

Shall we change this for 3.5.1 or leave it as is.

jazdw, 2018-10-04:
We will probably want to fix this, we should evaluate what it will break if someone doesn't update their UI module though.

terrypacker, 2018-10-09:
I just checked all the classes that extend EventTypeModel and they all have similar problems with returning Integer values instead of the ExportCode String.

jazdw, 2018-10-09:
I couldn't find any references to us using this in our front end code so it should be fine to change.