MangoAutomation/ma-core-public #1341: Rolling Backups - System Wide

terrypacker, 2018-10-10:
Disks are filling up with data due to endless backups being made.  All backups should implement a rolling strategy.

Puckfist, 2018-10-10:
Which do you want to point the finger at? All three backup types have a "versions to keep" setting. Or, is the premise that disks not filling up is foregone better than backups or logs existing (to the second to last backup!)? One could add a task that gets panicky at a given disk threshhold and sweeps up files it thinks are expendable. Maybe it needs to be explicitly configured with disks to watch and threshholds?

Maybe the issue is not copying the backups off the server and deleting them?

Puckfist, 2018-10-17:
I noticed on a MangoES in the field that the IDrive backup logs were filling up 1.7 GB. Didn't look like they were rolling over, since there were two years worth. The problem came about once there were tons of tsdb-corruption logs, which had a line in the backup log for each file. This meant every backup log was now MB instead of KB. Probably was mitigated by putting all the corruption logging into the same file, but might this have been what Jared saw?