MangoAutomation/ma-core-public #1344: H2 - LOB Expansion

terrypacker, 2018-10-15:
The LOB Expansion bug has been fixed in H2 but upgrading H2 is only possible by dumping the database to an SQL file and then re-importing.  

Some things to note:

1.  We currently use the PageStore, perhaps we want to change the MVStore since we will be exporting and re-importing anyway.  At present I'm not sure what the advantages are.

2.  If we do this we need to make sure that there is enough disk space to perform the upgrade on any given system.

H2 reference issues:
https://github.com/h2database/h2database/issues/1247#issuecomment-400924831
https://github.com/h2database/h2database/issues/301

Some code to show the growth:
https://github.com/infiniteautomation/h2lob-bug

Puckfist, 2018-10-19:
IIRC the issue is that you cannot open the PageStore database twice. So, maybe we can get away with only including the later version of H2 and taking the SQL dump in the first opening of the database. Of course, that's not a robust solution if something goes wrong, so we probably do need to bundle 196 into the upgrade too and be careful in our jar loading. Maybe we need to put 196 somewhere other than Mango/lib/ forever more?

Puckfist, 2018-11-02:
I think the solution is going to be to add a system settings entry for the latest version of H2 used to launch on the database, and assume that 196 is the right JAR to use if that entry is missing. i found that my 196 JAR was able to run selects and sql dumps on a 197 database. So, I think that'll be the way to key off any logic about what version the database is (test reading unknown databases with 196).

For the sake of mergrations / merges, I think the easiest thing to do will be to use 196 and not upgrade the database.
[h2ConversionNotes.txt](https://github.com/infiniteautomation/ma-core-public/files/2544142/h2ConversionNotes.txt)

Puckfist, 2019-01-23:
And then I think there'd be a env.properties entry to forego the 196 check step...

terrypacker, 2019-05-01:
I've taken a crack at this without including the 196 jar.  First off there is the concept of a `create version` in h2 which is stored in a table as per @Puckfist 's comment above.  We can use that to find out what version of h2 created the database.

The idea is that we can key off of 2 things to determine if we need to attempt a conversion.  A properties file (that we will create in this upgrade) that contains the create version of the database.  And if that file is missing we can check the create version via querying the database.

The reason we want the properties file is because we need to make a copy of the database before we open it because once we open a 196 or before db we can never open it again.

The logic to do the upgrade is roughly as follows:

1. Before doing anything like creating a db connection we check to see if there is a db file.  If not then abort because we are creating a new one, just make sure that when it is created we create the properties file with our current version in it.

2. If there is a database and the properties file DNE or has a version <= 196 proceed.

3. Check to see if there is an existing backup of the 196 database that was made in a previous upgrade attempt that failed.  If so copy that into the active db location, otherwise make a copy of the active db to save in case things go wrong.

4.  Make a connection and check the create version.  If the create version is <= 196 proceed. Else update properties file, delete copy and abort.

5. Drop the database as SQL statements into a zip file

6. close connection and delete active db.

7.  create new connection to fresh database that runs the dropped SQL on initialization

8.  select the create version and update properties file

9.  shutdown the connection and compact the db

10. clean up any backups/copies made.

terrypacker, 2019-05-01:
For SQL restore tasks I think we would need to do something similar and upgrade them safely before restoring.

This approach does use a lot of disk space but seems easier than trying to load and unload multiple h2 jars during runtime.

terrypacker, 2019-05-02:
When testing keep an eye out for the actual performance of this H2 version.  I had to modify the test framework as the in memory database behaved differently.  I couldn't really tell what was going on but the tables were missing occasionally.  The should be dropped and re-created between every test and somehow tests were being run and no tables were in the db...

closed 2019-05-10 by terrypacker