MangoAutomation/ma-core-public #1347: deletePointValuesBefore error , help me  please

190948804, 2018-10-29:
ERROR 2018-10-30T03:05:01,778 (DataPurge$DataPurgeTask.run:325) - Negative size 
java.lang.IllegalArgumentException: Negative size
	at sun.nio.ch.FileChannelImpl.map(FileChannelImpl.java:865) ~[?:1.8.0_181]
	at com.infiniteautomation.tsdb.impl.ChecksumMappedByteBufferInputStream.<init>(ChecksumMappedByteBufferInputStream.java:41) ~[ias-tsdb-1.3.8.jar:?]
	at com.infiniteautomation.tsdb.impl.ReversibleShard.openDataIn(ReversibleShard.java:250) ~[ias-tsdb-1.3.8.jar:?]
	at com.infiniteautomation.tsdb.impl.Shard.query(Shard.java:471) ~[ias-tsdb-1.3.8.jar:?]
	at com.infiniteautomation.tsdb.impl.CompressibleShard.query(CompressibleShard.java:391) ~[ias-tsdb-1.3.8.jar:?]
	at com.infiniteautomation.tsdb.impl.Series.query(Series.java:604) ~[ias-tsdb-1.3.8.jar:?]
	at com.infiniteautomation.tsdb.impl.IasTsdbImpl.count(IasTsdbImpl.java:1078) ~[ias-tsdb-1.3.8.jar:?]
	at com.infiniteautomation.nosql.InternalGenericDao.deleteItemsBefore(InternalGenericDao.java:236) ~[mangoNoSqlDatabase-3.4.1.jar:?]
	at com.infiniteautomation.nosql.MangoNoSqlPointValueDao.deletePointValuesBefore(MangoNoSqlPointValueDao.java:264) ~[mangoNoSqlDatabase-3.4.1.jar:?]
	at com.serotonin.m2m2.rt.RuntimeManagerImpl.purgeDataPointValues(RuntimeManagerImpl.java:911) ~[mango-3.4.5-M3.jar:?]
	at com.serotonin.m2m2.rt.maint.DataPurge.purgePoint(DataPurge.java:155) ~[mango-3.4.5-M3.jar:?]
	at com.serotonin.m2m2.rt.maint.DataPurge.executeImpl(DataPurge.java:82) ~[mango-3.4.5-M3.jar:?]
	at com.serotonin.m2m2.rt.maint.DataPurge.execute(DataPurge.java:64) ~[mango-3.4.5-M3.jar:?]
	at com.serotonin.m2m2.rt.maint.DataPurge$DataPurgeTask.run(DataPurge.java:323) [mango-3.4.5-M3.jar:?]
	at com.serotonin.timer.Task.runTask(Task.java:179) [mango-3.4.5-M3.jar:?]
	at com.serotonin.timer.TaskWrapper.run(TaskWrapper.java:23) [mango-3.4.5-M3.jar:?]
	at com.serotonin.timer.OrderedThreadPoolExecutor$OrderedTaskCollection.run(OrderedThreadPoolExecutor.java:336) [mango-3.4.5-M3.jar:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_181]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_181]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_181]

190948804, 2018-10-29:
Unable to delete data, disk consumption greatly

Puckfist, 2018-10-31:
Hi 190948804,

It is our preference to perform initial customer service on our forum, https://forum.infiniteautomation.com

This error is related to the size of the file and the stream type that is attempting to be used. You could get Mango started and run the purge you're trying to do by changing the env.properties entry,

db.nosql.shardStreamType=MAPPED_BYTE_BUFFER

to

db.nosql.shardStreamType=RANDOM_ACCESS_FILE

You may experience slower overall performance, but you could switch back once the necessary data was removed, or leave it there if you intend to be swelling shards to that size (I would guess over 2 GB) regularly.

Puckfist, 2018-10-31:
Your other option is to delete the large files in Mango/databases/mangoTSDB/*/* while Mango is not running.

Puckfist, 2018-11-02:
closing as support not development

closed 2018-11-02 by Puckfist

190948804, 2018-11-26:
Work too busy, forget to thank, thank you again