MangoAutomation/ma-core-public #1361: Virtual Serial Server Socket ip white list

Puckfist, 2018-11-16:
Because the virtual serial server socket only accepts one connection, it is possible for a port to get tied up by a malicious connection that doesn't close the connection. One solution would be firewall rules to prevent other machines from contacting that port than the intended. Mango could also provide this firewalling with the existing ip whitelist code.

Puckfist, 2018-11-16:
The backend is done, but the only way to change this property is via import/export JSON currently. Saving the port again through the UI will reset the ipWhiteLIst to null, which means to permit all IPs (the old behavior), which should probably be resolved before all is said and done.

terrypacker, 2020-07-16:
This is now working in the new UI

closed 2020-07-16 by terrypacker