MangoAutomation/ma-core-public #1378: User cannot use previous password

terrypacker, 2019-01-04:
Modify the User such that they cannot change their password to the existing password.

terrypacker, 2019-02-28:
@jazdw there is only 1 outstanding question of what to do when a pre-hashed password is supplied.  The code currently compares the passwords as encrypted using `Common.checkPassword(password, hash, true)` which I don't actually think works with BCrypt.

jazdw, 2019-02-28:
The user cannot supply a "pre-hashed password" when changing their password so I'm not sure what you are asking. `Common.checkPassword()` definitely does support BCrypt, why do you think that it does not?

terrypacker, 2019-02-28:
Prior to my changes yesterday I was not seeing any code that would prevent updating a user with a password of the form: `{BCRYPT}$2a$10$0qFsE4XNUPRoycA1eFikNOd3H` as it would not enter into any validation code.

And for `Common.checkPassword()`:

The point I'm making is that I am passing in true as the last parameter and passing something like the above example password with prefix and all.

Maybe I don't understand how that code is supposed to work but AFAIK you can't compare 2 separately hashed plain text values to see if they were the same plain text before the hash as is done on line 626.

If the rule is that one has to let Mango hash the password then I can change the validation code to enforce that.

jazdw, 2019-02-28:
> Prior to my changes yesterday I was not seeing any code that would prevent updating a user with a password of the form: `{BCRYPT}$2a$10$0qFsE4XNUPRoycA1eFikNOd3H` as it would not enter into any validation code.

Correct, it is not possible to validate a hashed password.

> The point I'm making is that I am passing in true as the last parameter and passing something like the above example password with prefix and all.

You should be passing in the plain text password that the user wishes to change their password to and their old hash from the DB.

closed 2019-04-09 by terrypacker

jazdw, 2019-04-11:
Corrections made in 9edb6d08a2c7e610253b51d62135d58dec1fd752 as per comments above