MangoAutomation/ma-core-public #1390: Change default behavior of empty permission

terrypacker, 2019-02-26:
Currently an empty permission implies superadmin permission is required.  This proposes to change the behavior so that an empty permission means anyone has access.

An upgrade must be done for All permissions to set anything that is empty to 'superadmin'.  Also change the Permissions class logic to match.

Review all VOs read/edit permission to ensure this will not be a problem.

Another viable option to investigate is if we change the 'user' permission to be a requirement of all users.  This would allow us to use it as the default permission which implies that everyone has access.

jazdw, 2019-02-28:
> Another viable option to investigate is if we change the 'user' permission to be a requirement of all users. This would allow us to use it as the default permission which implies that everyone has access.

Upon further reflection I think this may be the best option. It will be more seamless for our customers and will touch less code. Also just consider changing the Permissions class to always return true for a check if a valid user has the 'user' permission rather than doing a DB upgrade to add it to every user in the table.

jazdw, 2019-02-28:
Any change must be very clearly documented in the release notes.

terrypacker, 2019-03-01:
This has been added via 80d381b77b39b8d0e10c2dd0a49838566bac3d4b

closed 2019-04-09 by terrypacker