MangoAutomation/ma-core-public #1391: WebSocket exception thrown when logging in

jazdw, 2019-02-26:
Steps to reproduce
- Have browser open and logged into Mango
- Terminate mango
- Start mango back up
- Wait to be redirected back to login page
- Login

```
WARN  2019-02-26T13:33:11,190 (org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension$Flusher.failed:418) -  
org.eclipse.jetty.io.EofException: null
	at org.eclipse.jetty.io.ChannelEndPoint.flush(ChannelEndPoint.java:286) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.WriteFlusher.flush(WriteFlusher.java:393) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.WriteFlusher.write(WriteFlusher.java:277) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.AbstractEndPoint.write(AbstractEndPoint.java:380) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.io.FrameFlusher.flush(FrameFlusher.java:218) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.io.FrameFlusher.process(FrameFlusher.java:157) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.util.IteratingCallback.processing(IteratingCallback.java:241) ~[jetty-util-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.util.IteratingCallback.iterate(IteratingCallback.java:224) ~[jetty-util-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.io.FrameFlusher.enqueue(FrameFlusher.java:90) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.outgoingFrame(AbstractWebSocketConnection.java:495) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.AbstractExtension.nextOutgoingFrame(AbstractExtension.java:191) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.compress.PerMessageDeflateExtension.nextOutgoingFrame(PerMessageDeflateExtension.java:116) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension.access$600(CompressExtension.java:42) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension$Flusher.deflate(CompressExtension.java:447) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension$Flusher.process(CompressExtension.java:431) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.util.IteratingCallback.processing(IteratingCallback.java:241) ~[jetty-util-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.util.IteratingCallback.iterate(IteratingCallback.java:224) ~[jetty-util-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension.outgoingFrame(CompressExtension.java:218) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack$Flusher.process(ExtensionStack.java:419) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.util.IteratingCallback.processing(IteratingCallback.java:241) ~[jetty-util-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.util.IteratingCallback.iterate(IteratingCallback.java:224) ~[jetty-util-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.outgoingFrame(ExtensionStack.java:296) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.WebSocketRemoteEndpoint.uncheckedSendFrame(WebSocketRemoteEndpoint.java:311) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.WebSocketRemoteEndpoint.sendAsyncFrame(WebSocketRemoteEndpoint.java:241) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.WebSocketRemoteEndpoint.sendPing(WebSocketRemoteEndpoint.java:370) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketSession.sendPingMessage(JettyWebSocketSession.java:216) ~[spring-websocket-5.0.10.RELEASE.jar:5.0.10.RELEASE]
	at org.springframework.web.socket.adapter.AbstractWebSocketSession.sendMessage(AbstractWebSocketSession.java:111) ~[spring-websocket-5.0.10.RELEASE.jar:5.0.10.RELEASE]
	at com.serotonin.m2m2.web.mvc.rest.v1.websockets.MangoPingPongTracker.sendPing(MangoPingPongTracker.java:50) ~[?:?]
	at com.serotonin.m2m2.web.mvc.rest.v1.websockets.MangoPingPongTracker.<init>(MangoPingPongTracker.java:43) ~[?:?]
	at com.serotonin.m2m2.web.mvc.rest.v1.websockets.MangoWebSocketHandler.startPingPong(MangoWebSocketHandler.java:195) ~[?:?]
	at com.serotonin.m2m2.web.mvc.rest.v1.websockets.MangoWebSocketHandler.afterConnectionEstablished(MangoWebSocketHandler.java:169) ~[?:?]
	at com.serotonin.m2m2.web.mvc.rest.v1.websockets.MultiSessionWebSocketHandler.afterConnectionEstablished(MultiSessionWebSocketHandler.java:30) ~[?:?]
	at org.springframework.web.socket.handler.WebSocketHandlerDecorator.afterConnectionEstablished(WebSocketHandlerDecorator.java:70) ~[spring-websocket-5.0.10.RELEASE.jar:5.0.10.RELEASE]
	at org.springframework.web.socket.handler.LoggingWebSocketHandlerDecorator.afterConnectionEstablished(LoggingWebSocketHandlerDecorator.java:48) ~[spring-websocket-5.0.10.RELEASE.jar:5.0.10.RELEASE]
	at org.springframework.web.socket.handler.ExceptionWebSocketHandlerDecorator.afterConnectionEstablished(ExceptionWebSocketHandlerDecorator.java:48) ~[spring-websocket-5.0.10.RELEASE.jar:5.0.10.RELEASE]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketHandlerAdapter.onWebSocketConnect(JettyWebSocketHandlerAdapter.java:72) ~[spring-websocket-5.0.10.RELEASE.jar:5.0.10.RELEASE]
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:1.8.0_144]
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:1.8.0_144]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_144]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_144]
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onConnect(JettyAnnotatedEventDriver.java:148) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.openSession(AbstractEventDriver.java:233) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.WebSocketSession.open(WebSocketSession.java:616) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.WebSocketSession.onOpened(WebSocketSession.java:544) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.AbstractConnection.onOpened(AbstractConnection.java:209) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.AbstractConnection.onOpen(AbstractConnection.java:202) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onOpen(AbstractWebSocketConnection.java:446) ~[websocket-common-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.AbstractEndPoint.upgrade(AbstractEndPoint.java:440) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.server.HttpConnection.onCompleted(HttpConnection.java:385) ~[jetty-server-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.server.HttpChannel.onCompleted(HttpChannel.java:743) ~[jetty-server-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:510) ~[jetty-server-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:260) ~[jetty-server-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
[... cut ...]
	at sun.nio.ch.SocketDispatcher.write0(Native Method) ~[?:1.8.0_144]
	at sun.nio.ch.SocketDispatcher.write(SocketDispatcher.java:51) ~[?:1.8.0_144]
	at sun.nio.ch.IOUtil.writeFromNativeBuffer(IOUtil.java:93) ~[?:1.8.0_144]
	at sun.nio.ch.IOUtil.write(IOUtil.java:51) ~[?:1.8.0_144]
	at sun.nio.ch.SocketChannelImpl.write(SocketChannelImpl.java:471) ~[?:1.8.0_144]
	at org.eclipse.jetty.io.ChannelEndPoint.flush(ChannelEndPoint.java:264) ~[jetty-io-9.4.12.v20180830.jar:9.4.12.v20180830]
	... 62 more

```

Possibly related to #1356 and #1370

terrypacker, 2019-02-28:
I have found that I cannot reliably reproduce the issue using the steps above although it happens regularly while I am developing.

I have tried:

- navigating to specific pages and then following the steps above.
- restarting Mango 2x after shutting it down before logging in.

terrypacker, 2019-02-28:
It sounds crazy but if I compile/install a module while Mango is down I seem to reliably get the error.

terrypacker, 2019-04-03:
Also I see this happening for this web socket:

ws://mango.terrypacker.com:8081/rest/v1/websocket/json-data

terrypacker, 2019-04-03:
@jazdw I think that may be the problem, you are sending a subscription request to that websocket but it's a v1 socket that doesn't support that.  We have a v2 websocket for the JSONData I wonder if you use that this problem will go away?

jazdw, 2019-04-04:
I don't think its got anything to do with the subscription. AFAIK subscription is supported on the v1 socket. I can see however, that under the conditions outlined above, the UI tries to open a WS and it closes 90ms later.

terrypacker, 2019-04-04:
From my review the v1 JsonDataWebSocketHandler extends the  com.serotonin.m2m2.web.mvc.rest.v1.websockets.DaoNotificationWebSocketHandler which does not appear to support subscriptions.  The v2 JsonDataWebSocketHandler extends com.infiniteautomation.mango.rest.v2.websocket.dao.SubscriptionDaoWebSocketHandler.

There is no code in the v1 hierarchy that implements handleTextMessage(WebSocketSession session, TextMessage message) from what I can tell.

jazdw, 2019-04-04:
OK, I think I tracked it down. You're right it has to do with installing a module, if the UI detects a module has been installed/upgraded it reloads the page when logging in. Before it reloads the page it attempts to establish a new WS connection which is shortly thereafter destroyed.

jazdw, 2019-04-09:
This commit should fix this issue c153b437bf545b75a78010bec1a85d8dbcf3512d
It could still possibly happen if someone uses a login component on their own dashboard etc but I can't find a good solution for if a window is actually going to be unloaded or not (can't detect if they click OK or cancel).

closed 2019-04-09 by jazdw