MangoAutomation/ma-core-public #1392: Polling Data Source - Option to not poll at all

terrypacker, 2019-02-28:
We currently have 2 polling data sources that can be setup to not poll at all.  We could fairly easily make this an option for all polling style data sources. The use case would be to setup a Data Source that is only used when you set point values on it and it pushes them out i.e. Modbus.

Something to consider is that if a data source is not polling it would need special handling to be sure and update its PointsList during runtime.

jazdw, 2019-02-28:
UI fix for this - https://github.com/infiniteautomation/ma-dashboards/commit/5e9b27587d6064325fe35138231b7f9a661876a0

terrypacker, 2019-02-28:
There is also the consideration of enabling the Interval logging for each data point.  This is done in the Scripting data source for reference.