MangoAutomation/ma-core-public #1395: Emails may default to charsets with limited language reports

Puckfist, 2019-03-08:
A user from China observed the Chinese lettering in the "From name" of the email were converted into question marks. I found that if I defined the mail.mime.charset java option as UTF-8 the characters were properly transmitted. I did this via extension scripts like, 

set-mail-font.bat
if "%1" == "init" (
rem Set the mail charset to support more languages
set JAVAOPTS=%JAVAOPTS% -Dmail.mime.charset=UTF-8
)

set-mail-font.sh
#!/bin/bash
case "$1" in
init)
# Set the mail charset to support more languages
JAVAOPTS="$JAVAOPTS -Dmail.mime.charset=UTF-8"
;;
esac

terrypacker, 2019-04-09:
I think this can be fixed by setting the EmailContent encoding to UTF-8.  This happens in 2 places EmailWorkItem and ServerRestV2Controller.sendTestEmail.  In the former the setting is only applied if there is a custom template provided and the latter always uses UTF-8

Can you test that the test email works without setting the mail.mime.charset system setting?

Puckfist, 2019-04-30:
I'm not sure the instance description is in test emails, so I don't think I can supply test characters to the subject as required

Puckfist, 2019-06-05:
I put my "From name" as Chinese characters and got ?'s in the received email for the test email

Puckfist, 2019-06-05:
I don't think it has anything to do with the EmailContent's encoding setting.

Puckfist, 2019-06-05:
83c9a0e2a12b93f8ece23643872ab970139ee2cc

Puckfist, 2019-06-05:
Test email worked after my changes

closed 2019-07-10 by Puckfist