MangoAutomation/ma-core-public #1412: Events in the future?

Puckfist, 2019-04-18:
I observed with a no change detector that it was going active its duration in the future from updates it had just gotten!

This was on a system where the point was received over persistent TCP on logged events, with the source point logging on a five minute interval. The no change detector's interval was also 5 minutes, which lead to the client asking why they were getting so many events about not updating. I found through a scripting data source that the event detector reported its event was not active, while the event clearly showed as active on the events page.

Puckfist, 2019-04-18:
This is likely one thread getting into the pointData method and another in the scheduleTimeoutImpl to raise the event, so the pointData thread is updating the last change while the schedule timeout is raising the event with the last change as the base for the active timestamp

Simply passing the firetime from the timeout is probably insufficient, as that could still result in the event being raised when the pointData call was at the same millisecond and tried to cancel that task. So, maybe when raising the event the task need either check if its state is changed to cancelled or perhaps if the lastChange is at or after its fire time

terrypacker, 2019-04-18:
I believe this was mainly due to 2 problems in the DifferenceDetectorRT.  

The lastChange time was being set during the pointData() before the unscheduleJob() call was being made.  This would cause the detector to not raise and RTN an event in the past to show that there was actually no change during the time. 

I think it was assumed that if the pointData() method was called it would cancel the timeout task and not execute the scheduleTimeout method.  However there is a race condition that if the timeout task is actually running pointData() cannot unschedule it (this can only happen before the task executes).  So I have added code in the scheduleTimeout method to check that it should raise an event.

terrypacker, 2019-04-18:
I suspect that many of the detectors are not using the correct time when raising the events.  It seems some are using the timestamp of the point value when I presume they should be using the Timer time.

closed 2019-04-29 by terrypacker